andrivet/ADVi3pp #148: 3.0.2 Reports LCD Version 3.0.1

AddisonBurn, 2018-06-18:
### Bug Report

- [ ] You have checked that you are not creating a **duplicate** of an existing (maybe closed) issue (use the search)
- [ ] You have tested with the **latest** version of ADVi3++ (including Beta releases)
- [ ] You are able to reproduce the bug
- [ ] You are able to describe step by step the process to reproduce the bug

Flashed version 3.0.2 using the .img file for the LCD and the .hex file for the main board.
In the menu Info>Versions it reports main board version 3.0.2 and LCD version 3.0.1
I double checked and I definitely used ADVi3pp-LCD-3.0.2.img

andrivet, 2018-06-18:
You're right. I forget to increase the version of the LCD. So this is a bug but I will not fix it. Will be fixed in a later version

closed 2018-06-18 by andrivet

AddisonBurn, 2018-06-18:
I figured as much! Just letting you know for future versions.
Great work, the software is fantastic.

andrivet, 2018-08-11:
Fixed in 3.0.2 re-release.