andrivet/ADVi3pp #151: Restore Autolevel after running G29

brianhealey, 2018-06-24:
Thank you for submitting your feedback to the ADVi3++ project.

Please be sure to review the checklists below before submitting and use one of the templates. 

**It is mandatory**. If you do not comply with these rules, I will **delete** your issue.

- [ ] This issue is **NOT** about support. This is not the right place for support (there is an [ADVI3++ Facebook page](https://www.facebook.com/advi3pp/) for that)

------------------------------------------------------------
### Feature Request

- [ ] Your request is not specific to your setup but can benefit to many people
- [ ] You have checked that you are not creating a **duplicate** of an existing (maybe closed) request (use the search)
- [ ] You have provided a short but meaningful title
- [ ] You have provided enough information (such as links to web sites, blogs, technical description) so that anyone can understand what you are talking about. Just any only a title is not enough.
- [ ] You have provided objective **facts**, not your personal feelings, your personal preferences.

Note: It is no more necessary to add "[FR]" in the title. You can, if you want, label your issue with "enhancement".

Detailed description of the feature request:

_Please provide information (such as links to web sites, blogs, technical description)._

The 1.1.x bugfix branch contains a `#define RESTORE_LEVELING_AFTER_G28` feature that I would love to see integrated into the mainline ADVi3pp release. 

------------------------------------------------------------
### Bug Report

- [ ] You have checked that you are not creating a **duplicate** of an existing (maybe closed) issue (use the search)
- [ ] You have tested with the **latest** version of ADVi3++ (including Beta releases)
- [ ] You are able to reproduce the bug
- [ ] You are able to describe step by step the process to reproduce the bug

* Exact model of your **printer**:
* **ADVi3++** exact version:
* Printing using **SD card** or **USB**:
* **For USB printing only** Software used to print with its exact version:

#### Description of the bug

* Expected behaviour:
* Actual behaviour:
* Steps to reproduce:
  - Do this
  - Do that

tobashadow, 2018-06-24:
As much as i enjoy adding features to my fork to test, honestly i would just add a G29 to your start gcode for every file. you could bump the bed screws taking off a print or even loading filament. Having it take the three or so minutes to run a fresh bed leveling mesh right before it starts is worth it if your doing a all day print.

brianhealey, 2018-06-24:
This is already in the 1.1.x-bugfix branch for marlin.  I integrated it into my fork as well and like that I don’t have to perform a G29 repeatedly when I am having to restart a failed print.

andrivet, 2019-05-26:
Implemented in version 4.0

closed 2019-05-26 by andrivet