andrivet/ADVi3pp #153: [FR] Graph as default view

FierceBun, 2018-06-24:
Many folx use remote systems to operate thier printer with minimal physical interaction.  The most useful screen from across the room is the activity graph.  Either an option to tick to make it the default screen, or a command that can be sent along with gcode to open the graph screen would facilitate this usage.

jmdefino, 2018-06-26:
I was going to suggest the same. The original firmware had a progress bar. This was very useful as you could see it from across the room instead of having to get right to the screen to view the percentage completed. Thanks for your consideration.

LordJakson, 2018-06-27:
I thought about the same. With the command M24 (Start SD print) the display should swich to the Temp Graph so I can start it remoly.

andrivet, 2022-10-30:
Implemented in version 5.5.0. Implicates major refactoring of code.

closed 2022-10-30 by andrivet