andrivet/ADVi3pp #155: Support Wanhao Duplicator 9 (D9)

andrivet, 2018-06-28:
Thank you for submitting your feedback to the ADVi3++ project.

Please be sure to review the checklists below before submitting and use one of the templates. 

**It is mandatory**. If you do not comply with these rules, I will **delete** your issue.

- [x] This issue is **NOT** about support. This is not the right place for support (there is an [ADVI3++ Facebook page](https://www.facebook.com/advi3pp/) for that)

------------------------------------------------------------
### Feature Request

- [x] Your request is not specific to your setup but can benefit to many people
- [x] You have checked that you are not creating a **duplicate** of an existing (maybe closed) request (use the search)
- [x] You have provided a short but meaningful title
- [x] You have provided enough information (such as links to web sites, blogs, technical description) so that anyone can understand what you are talking about. Just any only a title is not enough.
- [x] You have provided objective **facts**, not your personal feelings, your personal preferences.

Note: It is no more necessary to add "[FR]" in the title. You can, if you want, label your issue with "enhancement".

#### Detailed description of the feature request:

The [Wanhao Duplicator 9 (D9)](http://www.wanhao3dprinter.com/Unboxin/ShowArticle.asp?ArticleID=92) hardware is very close to the hardware of the Duplicator i3 Plus Mark II. In particular, it uses the same kind of LCD panel. It seems thus possible to support this printer.

andrivet, 2018-07-20:
Sources: 
https://github.com/JeremyProffitt/Marlin-Wanhao-D9-300

taxilian, 2018-10-24:
Updated source: https://github.com/garychen99/D9/blob/master/D9%20300%20V0.164(B).zip

I would be interested in helping with this if possible; I have experience building marlin from source, configuring for a new printer, and some experience with arduino development; I'd need help and pointers, but I'm willing to contribute some time, definitely can test, and can hopefully help with some code if someone can point me in the right direction. I'm also willing to do a patreon donation if it'll help this happen but I'd like to know what is going on before committing financially =]

CameronLamont, 2019-06-05:
I downloaded the avdi3pp release and modified the config best I could based on th d9 mk2 firmware. I slso flashed the lcd firmware. This included a new board/pinout which is very close to i3

Printer started up but wouldn't follow commands from serial or lcd. It slso was resetting almost immediately after starting up (based on serial messages). My theory at present is the power break tripping something but thst is a guess. 

I had previously attempted to graft the d9 fork onto vanilla Marlin 1.1.9 but as yet have not got it compiling. Am thinking orking advi3pp would be a better starting point.  I haven't been able to successfully load the dwin project from wanhao, need to lok at directly downloading from the lcd or going from scratch / advi3 baseline.

andrivet, 2019-06-05:
@CameronLamont Is your post a support request? A bug report? Some technical information? It does not look like it is. This place is not the right one for support request so I will delete your comment.

ryamoo, 2019-06-05:
It looked like he was trying to contribute some information to an open issue that doesn't seem to have had any progress since it was opened 11 months ago due to the 4.0 work (understandably), but perhaps he shouldn't have bothered

andrivet, 2019-06-05:
@ryamoo He is not (he gives no technical information). I try to keep the issues as clean as possible. The rules are in the template of issues. This issue (created my me) has no progress because it was not voted by the supporters of the project. And if I am not wrong, other people have successfully used ADVi3++ (modified) for D9.

CameronLamont, 2019-06-05:
My apologies, it was intended to be constructive. I will search for those
who have made the conversion.

taxilian, 2019-06-05:
I thought it was useful information and very appropriate to the post,
particularly given that Andrivet has indicated that this is a low priority
issue. I was more than a little disappointed that it was deleted.

I understand the intent, but in my unsolicited opinion removing it was
unnecessary and counter-productive to the goal of advancing an open source
project, which this theoretically is.

I understand about keeping issues clean, but I thought it was very concise:
it stated exactly what he tried and what the result was. It did not whine
about the problems, nor did it beg for help -- actually it just indicated
what he planned to do next to try to work on the issue.

In some circles this is called collaboration, but I have to agree
with @ryamoo that the reaction from the repo owner to me says "no, don't
work on this!". One reason I like this project is that it is open source,
but open source requires collaboration and discussion -- if discussion is
shut down before it can start it pretty much ensures that no productive
work can be done by anyone else.

Not gunna lie... this has me rethinking my patreon contributions.

- Richard a.k.a. taxilian

andrivet, 2019-06-05:
@taxilian Everything is explained in the the template of the issue ("If you do not comply with these rules, I will **delete** your issue.") . This is my repository, my rules. There are other places to make such discussion. Just not here, please. If you prefer other rules, you are free to make a fork of this project.

taxilian, 2019-06-05:
Your repository, your rules.

My money, I choose what to support. Your rules are counter-productive to
the good of the project and not something I can support. I wish you well in
the future and genuinely appreciate what you've done on this -- but I
really hope that you'll reconsider this approach since it basically
entirely shuts down any third party involvement in the development, which
is not the type of project that I can support.

On the bright side, it's only a small amount from any one person, but it is
a small amount that I will put towards another creator.

Good luck in the future.

Richard

ryamoo, 2019-06-05:
I thought those rules were for creating new issues, not about contributing comments to an open issue to help with the discussion.
Perhaps you need to update those rules to make it more clear just how draconian you want to be with your repo.

Split7fire, 2019-06-06:
Trying to adapt ADVi3pp to work with WanHao D9 400. Successfully flashed LCD part and Marlin (with minor changes to bed sizes, probe conf, steps/per/mm etc). But when printer starts, I see animation (rotating marlin icon) and then screen changes to main mode (temp data, 5 touchscreen buttons etc), but seems that no connection between mainboard and lcd was made. Everything that I got is temperature readings (seems right, numbers changing if I heat probes manually by hand), but in position where version number should be placed some corrupted symbol displayed. Also while watching marlin output from USB port, seems that marlin reboots every 1 sec (I notice repeated "start" sequence). I tried version 3 and 4(beta) - same result.  Any help would be awesome.

andrivet, 2019-06-06:
@Split7fire No support here.

Split7fire, 2019-06-06:
@andrivet, so where one can get support?

andrivet, 2019-06-06:
@Split7fire until i release version 4, only here: https://www.patreon.com/andrivet