andrivet/ADVi3pp #156: Filament load/unload artifacts

chschmit, 2018-07-03:
### Bug Report
Wanhao i3 Duplicator Plus
ADVi3 3.0.2
print from SD card

#### Description of the bug
Expected behaviour: loading/unloading comments shown in the middle of the screen.
Actual behaviour: loading/unloading comments are shown but includes the previously printed filename 
after the comments which messes up the comment text...word wrapping issue.

* Steps to reproduce:
This only happens after something was printed and then choose to load/unload filament. If the printer is restarted, the issue isn't there.

DynaMight1124, 2018-07-21:
[image]

Heres a picture of the issue

andrivet, 2019-05-26:
Fixed in version 4.0

closed 2019-05-26 by andrivet