andrivet/ADVi3pp #162: Remove XYZ Motors "calibration"

andrivet, 2018-08-05:
- [X] This issue is **NOT** about support. This is not the right place for support (there is an [ADVI3++ Facebook page](https://www.facebook.com/advi3pp/) for that)

------------------------------------------------------------
### Feature Request

- [X] Your request is not specific to your setup but can benefit to many people
- [X] You have checked that you are not creating a **duplicate** of an existing (maybe closed) request (use the search)
- [X] You have provided a short but meaningful title
- [X] You have provided enough information (such as links to web sites, blogs, technical description) so that anyone can understand what you are talking about. Just any only a title is not enough.
- [X] You have provided objective **facts**, not your personal feelings, your personal preferences.

#### Detailed description of the feature request:

I know this will create debates, but I have decided to remove the **XYZ Motors Tuning**:

[image]

Why? Because it is wrong. It gives the false impression that this part needs some calibration when in fact, these parameters (number of steps per millimeter) are directly derived from the mechanical parts of the printer (the stepper motor, ...) Unless you change these parts (like the motors), there is no need to change these parameters. The default values are `80`, `80`, and `400`.

The maths are the following:

**X and Y axis**

- The motors moves `1.8`&deg; per steps
- So a revolution has `360 / 1.8  = 200` steps
- The driver gives `16` micro-steps
- On X and Y axis, the toothed pulleys have `20` teeths
- The belts have a `2` mm tooth pitch
- So we have: `(200 * 16) / (20 x 2) = 80` steps / mm

**Z axis**

- The motors moves `1.8`&deg; per steps
- The lead screw is `8` mm per revolution
- So we have: `(200 * 16) / 8 = 400` steps / mm

Some people objects that the mechanical parts of the printer are not perfect so that this tuning is still useful. In my point of view, this is wrong: the number of steps per revolution of a motor is precise enough, the number of teeths is a fix number (and the length or shape of each teeth has no importance here).

Another argument that is wrong in my point of view: You have precisely "calibrate" the X, Y and Z axis to print perfectly a 20x20x20 cube. And your numbers are slightly different to the defaults (`80`, `80`, and `400`). So you do not understand why I will remove this tuning. Try to print a bigger shape. like a 100x100x2 square. You will see that it is not right.

Even if I remove this part, you can still change yourself the parameters for each axis. Just do the math yourself:

```
new_steps_value = old_steps_value * expected / measured
```

So if we assume you currently have the default values and you print a 20x20x20 cube, it gives:

```
new_steps_x = 1600 / measured_x
new_steps_y = 1600 / measured_y
new_steps_z = 8000 / measured_z
```

------------------------------------------------------------

If you have strong arguments against this decision, do not hesitate to comment. But **PLEASE**, give **strong argument**, not just "I think", "I prefer", etc. **I will delete comments like that.**

Adrian57, 2018-08-05:
No disagreement here! You are absolutely correct.

bleughb, 2018-08-05:
Some disagreement here. It’s a very useful feature for those installing this firmware for the first time...many don’t know their steps per mm.. setting so this is great to calibrate the printer for the first time, or with new firmware....suggest to not remove it, but to move it to a ‘first time installed’ initial setup type screen that runs the first time you install the firmware or reset to factory defaults.......people that tweak firmware tweak hardware also! Empower them a little! ....also, many printer makers (wanhao, creality) have this setting available in menus! And, finally, for belt driven printers, this steps per mm does change with time as belts become worn! It takes a LONG time mind

andrivet, 2018-08-05:
@bleughb 
1) If you install the firmware (i.e. ADVi3++) for the first time, ADVi3++ will reset the settings (since they are not compatible with the old one) and thus it will correctly set the default values. So what you explain never happens
2) I never said (on the contrary) that I will remove the settings (Motors settings). You made a confusion between two different screens
3) "as belts become worn". If belts are too old, you have to change them. But your argument makes sense. I will think about it.

bleughb, 2018-08-05:
Agree totally about the 20mm cube not being good enough. Maybe change the page to suggest ‘print a 100mm or 150mm (or whatever) square....and ask for the measurements!?

Here’s a small blog I did about 20mm cubes...

https://bleughbleugh.wordpress.com/2018/07/25/calibration-wyisyg/

ryamoo, 2018-10-15:
I'd be in favour of the cube remaining, or something similar to what @bleughb suggests being added.
My prints were regularly too big on the Y Axis, and I never knew about the motor steps settings or even what they were. 
It wasn't until I installed ADVi3++ and saw the xyz calibration cube option, that I read into it more. I printed a cube, it was off, I entered the actual sizes and my prints have been near-perfect ever since. 

So if the 20x20x20 screen is disappearing, it would be good to have a "hint" on the motor steps setting screen (maybe a (i)-info button to pop-up more info) to suggest people look it up, because it was a massive help to me.

andrivet, 2018-12-10:
Implemented in Alpha 1

closed 2018-12-10 by andrivet

eages, 2020-04-19:
Hi Sebastien,
I know this issue has been closed for a while now but I'd like to offer a scenario that would be a valid argument for keeping these parameters. Please refer to the attached diagram representing the X axis interface to the print head. Here the linear rails are assumed to be parallel but the belt is not necessarily parallel to the rails hence small angles A1 and A2 exist between the belt-tension direction and the rails. A1 and A2 aren't necessarily equal and vary with head position.

[image]

For the simplest case, let's assume A1 = A2 near the center of the bed and belt tensions are nominal. If the stepper rotates in order to pull the belt 1 unit leftward (actually in the tangent direction to its pulley), the head will actually move approximately cos(A1) units. Further, some strain has been added to the belt and the print head now experiences a new, counterclockwise static moment which is resisted by the rails. When the stepper motion is reversed, the travel and torsion are reversed.

A similar argument applies when A1 = -A2, only a vertical reaction force is introduced to the rails.

For a local (approximately linear) region near the starting point, any combination of A1 and A2 may be represented as the superposition of the offset case (A1 = -A2) and the misalignment case (A1 = A2).

When an adjustment parameter for the X axis is present, inaccuracy in the neighborhood of calibration surfaces can be minimized. This can be useful if a particular feature requires a high degree of accuracy in the region of said surfaces.

In general, a multi-point calibration across the whole of the build plate would provide better distribution and reduction of error but the single value calibration is simpler to implement and benefits simple geometry.

Cheers.

andrivet, 2020-04-19:
Hi,
Interesting, but frankly it is just a theory without any estimation of the effects you described and, if I understand well, without experimentations.
Of course, the belt is not perfectly parallel, but several other things on the printer are not perfect. So what is your estimation of the A1 angle? On my printer, I am not able to measure any noticeable deviation from 0°. If it is higher, it is because the belt is not tight enough.

I will not reintroduced something fondamentally wrong based only on a theory. Sorry, but I need numbers and facts to be convinced.