andrivet/ADVi3pp #163: Title of last print stuck on LCD screen

Deose, 2018-08-05:
### Bug Report

- [Y] You have checked that you are not creating a **duplicate** of an existing (maybe closed) issue (use the search)
- [Y] You have tested with the **latest** version of ADVi3++ (including Beta releases)
- [Y] You are able to reproduce the bug
- [Y] You are able to describe step by step the process to reproduce the bug

* Exact model of your **printer**:
* **ADVi3++** exact version: ADVi3pp-BLTouch-Mainboard-3.0.2.hex
* Printing using **SD card** or **USB**: SD

#### Description of the bug

* Expected behaviour: Text on screen shouldn't have file name outside of printing that file or file selection
* Actual behaviour: At the end of different lines describing which action is happening, the name of the last file printed is added.
* Steps to reproduce:
  - Print something via SD card. 
  - Stop the print before it finishes.
  - Do something else (e.g. BLTouch Leveling, loading filament, etc.)
[image]
[image]

chschmit, 2018-08-07:
I'd identified this in Issue #156

andrivet, 2018-08-07:
@chschmit Yes, but this issue is more general. This is why I have not marked it as duplicate.

andrivet, 2019-05-26:
Fixed in version 4.0

closed 2019-05-26 by andrivet