andrivet/ADVi3pp #164: IPv4 displayed vs. IPv6

TTVert, 2018-08-16:
I saw no option to configure this . The IPv6 is displayed but can IPv4 be displayed instead somehow?

Thanks
Dave

andrivet, 2018-08-18:
@TTVert I have no idea of what you are talking about. This printer has no network. And please, use the template when submitting issues

TTVert, 2018-08-18:
You seriously don't know that at the bottom of the display on your firmware there is an IP v6 address displayed?

andrivet, 2018-08-18:
I assume that you are using OctoPrint, isn't it? This IP is displayed by OctoPrint. Nothing to do with the firmware.

closed 2018-08-18 by andrivet

TTVert, 2018-08-18:
I use repetier.  I never saw an option for it until after the latest firmware of yours.  I will go through all the repetier docs and settings and see where that may be coming from then.   It is displaying the ip v6 at the bottom of the screen while idle w/ no print running.  Unsure how repetier would be doing that but I'll look.

Dave

andrivet, 2018-08-18:
This is sent by Repetier. Again, the printer has no ethernet connector so it has no notion of IP address. What you see is a message sent by Repetier with the G-Code M117. The IP is the IP of your computer or whatever device you are using with an Ethernet connector.

I stop the conversation here since it is not about an issue but about support.

TTVert, 2018-08-18:
Fair enough, where is the best place to obtain feature info/support when needed?  i think I saw a paid facebook group?

andrivet, 2018-08-18:
Patreon. https://www.patreon.com/andrivet