andrivet/ADVi3pp #166: Bltouch + glass bed?

riaqn, 2018-08-30:
------------------------------------------------------------
### Feature Request

- [*] Your request is not specific to your setup but can benefit to many people
- [*] You have checked that you are not creating a **duplicate** of an existing (maybe closed) request (use the search)
- [*] You have provided a short but meaningful title
- [*] You have provided enough information (such as links to web sites, blogs, technical description) so that anyone can understand what you are talking about. Just any only a title is not enough.
- [*] You have provided objective **facts**, not your personal feelings, your personal preferences.

Note: It is no more necessary to add "[FR]" in the title. You can, if you want, label your issue with "enhancement".

Detailed description of the feature request:

Currently the BLtouch auto leveling samples 9 points on the bed. This is quite good for plastic bed sheet which is not completely flat, but not necessarily good for glass bed which is very very flat. Therefore, instead of using 9 samples to calculate the gradient of each area in bed, for glass bed it might be better to combine the 9 samples and calculate the most accurate gradient for the whole bed.

andrivet, 2018-08-30:
I do not see your point? Why an average is better than a mesh even if it is (relatively) flat? Why is it an advantage?

riaqn, 2018-09-07:
Hello, Marlin actually provides more than one options for auto leveling, check out here:
https://github.com/andrivet/ADVi3pp-Marlin/blob/a89543b609e983061155bd7ba09fa1a169a8fdbc/Marlin/Configuration.h#L900

My request is to add an option to allow users to choose from the two options, LINEAR and BILINEAR.

andrivet, 2018-09-07:
You really think that I am not aware of that? Why are you asking this ("to add an option to allow users to choose from the two options, LINEAR and BILINEAR"). For me, it does not make sense.

andrivet, 2018-09-07:
The goal of ADVi3++ is not to add all possible options, every possibility offered by Marlin. Marlin takes the assumption that you choose these options at compile time. So enabling 2 (non-compatible) options at compile time and selecting one or the other at runtime is complex. So unless there is a very good reason to have such options, I will not implement such complex code. If you have facts (such as a link to a document or post explaining why we need those options), I can revise my judgment. But for the moment, I will not implement what you asked. Of course, you can also make a fork and submit a Push Request.

riaqn, 2018-09-07:
I think you are right - I didn't realize how complex it is to enable both compile-time options at the same time.
If so, would you be so kind as to release two versions of BLtouch-enabled firmware? There will be `bltouch-glass` and `bltouch-plastic`.

If you are too busy to do so, I'd like some input from you because I'm trying to tinker the code and simply changing that macro definition seems not enough. Building it gives the following error:
```
Marlin/advi3pp.cpp: In member function 'void advi3pp::internals::Printer_::sensor_grid_show()':
Marlin/advi3pp.cpp:2308:50: error: 'z_values' was not declared in this scope
frame << Uint16(static_cast<int16_t>(z_values[x][y] * 1000));
^
```
which is obviously an auto-leveling related issue.

andrivet, 2018-09-07:
For your 1st question: another .hex means another firmware to flash, test, etc. when I am doing a new release. It means also more complexity for users (I already get support questions because there are 2 .hex files). So no, unless there is a good reason.

For your 2nd question: It is a support question. Sorry, but I am doing support only on Patreon.

riaqn, 2018-09-23:
> With a glas bed:
> An average of all 9 points makes only sense if the bed is complete level.
> To adjust the nozzle height.
>
> The only way to reduce the leveling time and probably reducing cpu load is
> to reduce the amount of points to 3 (to describe a plane) if you assume
> that your glasbed is flat and not level.

Obviously when I say "the average of nine points" I don't mean it
literally. From the nine points we can derive several triangles whose
vertices are among the nine points. The slope rate of the bed can then be
calculated to be the average slope rate of the triangles.

riaqn, 2018-09-25:
In case anyone is interested, here is the firmware(corresponds to 3.0.2).
[firmware.hex.gz](https://github.com/andrivet/ADVi3pp-Marlin/files/2413501/firmware.hex.gz)

Note that in you use the UI of leveling, it's gonna report all zeros, but actually it's working well.

andrivet, 2018-09-26:
@riaqn Please stop abusing this issue, this repository, my name and the name ADVi3++. Use our own repository and issues. Moreover, If are creating a fork and distribute some binary, you **have to** to publish your source code. What you are currently doing is a violation of the GPLv3 license. So please take the necessary steps ASAP otherwise, I will ban you.

riaqn, 2018-09-26:
@andrewsil1 thanks for the kind reminder. I have published the source code here: https://github.com/riaqn/ADVi3pp-Marlink-glassbed.

As for this issue: I feel it makes sense to show people why the original author thinks it's a bad idea, or why he didn't implement it, so if you don't mind I might link to this page from my repo's README.

riaqn, 2018-09-26:
BTW, since you said not having enough time is why you didn't implement it, would it help if I make the patch compatible with the original repo so you can merge it?

andrivet, 2018-09-26:
@riaqn No, I am not interested by your fork. In my point of view, it is useless. I also ask you again to stop abusing this **closed** issue. For example, use your own repo to distribute your binaries. Thanks.