andrivet/ADVi3pp #167: Blobs in corners of the prints

runam0K, 2018-09-11:
### Bug Report

* Exact model of your **printer**: Wanhao i3 Plus
* **ADVi3++** exact version: 3.0.2
* Printing using **SD card** or **USB**: USB
* **For USB printing only** Software used to print with its exact version: Octoprint 1.3.9

#### Description of the bug

Since the upgrade to ADVi3 ++ I have the problem that the printed objects have blobs at the corners. The edges are not straight. I have attached some pictures.

I'm not the only one with this problem either. My friend has exactly the same problems since switching to ADVi3++.
He tried all settings in the slicer several times. e.g. retraction, coasting distance.

The slicer is Simplify3D in both cases.

How can the problem be solved? Why did the problem not happen with the default firmware?

[image]
[image]
[image]
[image]
[image]
[image]
[image]

andrivet, 2018-09-11:
Please do not abuse GitHub issues with support requests. This is not the right place.

closed 2018-09-11 by andrivet

runam0K, 2018-09-11:
So where is the right place for problems with this firmware if not here?

andrivet, 2018-09-11:
In the template of issues, it is clearly stated: "This issue is **NOT** about support". You may get support from the community on the Facebook page, and from myself on Patreon.