andrivet/ADVi3pp #170: Disable drive when stopping print

multiple1, 2018-09-23:
### Feature Request
- [x] Your request is not specific to your setup but can benefit to many people
- [x] You have checked that you are not creating a **duplicate** of an existing (maybe closed) request (use the search)
- [x] You have provided a short but meaningful title
- [x] You have provided enough information (such as links to web sites, blogs, technical description) so that anyone can understand what you are talking about. Just any only a title is not enough.
- [x] You have provided objective **facts**, not your personal feelings, your personal preferences.

Detailed description of the feature request:

When a print finishes the ending script of the gcode usually lifts the head and then disables the motor drive.
When i monitor a failure while printing (e.g. not sticking to printbed) i press on the stop button. But the motors keep engaged, so i can not move away the printhead with my hands. I have to exit the print menu enter the move menu and disable drive. And during that time the head cools of and the nozzele melts a blob into the failed print.

Would it be possible to disable drive after pressing the stop button and optionally lift the nozzle (for 1cm)? I dont see any point to keep the motors enabled after stopping a print.

andrivet, 2019-05-26:
Implemented in version 4.0

closed 2019-05-26 by andrivet

multiple1, 2019-07-25:
Thanks for picking up my request.
But your missuse of Github with your Patreon strategy SUCKS REALLY BAD! 
👎

andrivet, 2019-07-26:
@multiple1 There is no misuse of GitHub. Apparently you do not understand what “open source” is. Open source does not mean open binaries.
On your side, you are not respecting the rules of this repository (no swearing)