andrivet/ADVi3pp #173: Print time + expected time left on print page

haydendonald, 2018-10-02:
Hi just a feature request.
Could it be possible to add a few statistic information on the print page such as expected print time, the total print time (so far), and a larger %? It'd be very useful (more useful than a temp graph in my opinion) :)

Another feature that'd be cool (probably not possible on the i3 but..) would be a power recovery feature, you might be able to do this by writing the current file + gcode line in memory and then doing a check. Im not sure how marlin works fully but that'd be sick :D

Thanks heaps, i've been using it for a bit now and it's very useful!

andrivet, 2019-05-26:
Issue is vague, and mix two different requests:
- I assume it is about SD printing. The firmware does not have the required information (and power) to output such information.
- Power recovery may be implemented in a future release (at least for Mark II printers).

closed 2019-05-26 by andrivet

DarthMuppet54, 2020-04-15:
Slight modification to suggestion - could *any* aspect of time be incorporated on the 'printing' page? I understand why you can't work out how long it has to go, but elapsed time is already being recorded, as you show it at the end. Could the time be shown and updated every minute? the slicer (well, Cura does) tells you how long it's going to take to print, but when you're busy, it's easy to forget when you started, and have no idea when your 5 hour print will complete. Thanks.