andrivet/ADVi3pp #174: LCD Touch panel support

ModMike, 2018-10-03:
I have a Creatbot DX that uses the same touch panel as the Wanhao.  Do you guys support the touch panel and can I port the settings to Marlin 2.0.x?

Thank you.

churd83, 2018-10-04:
Marlin 2.0.x is for 32bit only from my understanding

ModMike, 2018-10-04:
It works on both 32 and 8 bit.  I have it running on 2 different 8 bit board and 3 32bit boards.

andrivet, 2018-10-04:
I do not plan to support Creatbox printers

closed 2018-10-04 by andrivet

ModMike, 2018-10-04:
I didn’t ask you to support Creatbot, I asked you if you if you could support the Dwin touch panel from the Wanhao printer.

How do I configure it so Marlin sees it?  I get that it’s. a serial device acting as a host.  Are you really unwilling to share that knowledge and help the community at large appreciate your work?

andrivet, 2018-10-04:
@ModMike Do I owe you something? Are you assuming that I have an infinite amount of time to explain everything? I already share a lot (for example: http://sebastien.andrivet.com/en/posts/wanhao-duplicator-i3-plus-3d-printer/)

ModMike, 2018-10-04:
Vraiment?  Très chic comme réponse.

andrivet, 2018-10-04:
Closing, delete comments because the question was not related to this project.