andrivet/ADVi3pp #175: Progress visualisation method - enhancement

welshsijones, 2018-10-11:
I'm not sure if this is possible but if the x location, z location and and extruder status are known then an image could be maintained as the print progresses.  
The data could be kept in a binary array and an image could be built when displayed.

andrivet, 2018-10-11:
I do not understand what you suggest. An image of what? What for? And please, follow the template when creating an issue. It is mandatory. There are required element: "You have provided enough information (such as links to web sites, blogs, technical description) so that anyone can understand what you are talking about. Just any only a title is not enough"

bleughb, 2018-10-11:
I'm no advanced coder, But
I Can't understand how the atmega could have the processing power to 'slice' the model to display an arbitary 2 dimensional cut of it.....3D printers have no understanding of where the head is in space...it just moves to an arbitary place where the G-Code sends it.....
even if you did a quick 'boundary' calculation to find X min / x max and plotted that, it wouldn't represent in any real way the thing you're printing!

Why not spin the request into a 'layer number being printed' display - plenty of ways I can see that as being useful....just count the Z hops maybe?
and possibly a 'progress' per layer - 'may' be able to be done by counting the number of G code commands between Z hops and normalising to 100%?

Maybe that already exists? I'm just trying to help

ryamoo, 2018-10-11:
It wouldn't slice it, but based on gcode and bed size, the processor would know that a move from previous position of X107 to the new position of X135 combined with the extrude command, there is now printed filament in that space.
I think it's possible but I'm not sure why its needed really, because you'd usually know what it is you're printing and you can see the progress as it's printed. 
If it's for progress, maybe a % would be enough?
However when used with something like octoprint, the printer wouldn't know how long to go because octoprint parsers the gcode and sends it to the printer line by line as its needed.

andrivet, 2018-10-11:
@welshsijones Why do you think it will provide any value to the majority of the users? Is "would look great" the only reason? If so, no I will not implement something useless. So I consider it as a duplicate of #27.

closed 2018-10-11 by andrivet