andrivet/ADVi3pp #177: Cancel e-steps tuning

andrivet, 2018-10-13:
Thank you for submitting your feedback to the ADVi3++ project.

Please be sure to review the checklists below before submitting and use one of the templates. 

**It is mandatory**. If you do not comply with these rules, I will **delete** your issue.

- [x] This issue is **NOT** about support. **This is not the right place to ask for support**. You **may** get support from the [ADVI3++ Facebook page](https://www.facebook.com/advi3pp/). You can get support from the developer of ADVi3++ from its [Patreon page](https://www.patreon.com/andrivet).

------------------------------------------------------------
### Feature Request

- [x] Your request is not specific to your setup but can benefit to many people
- [x] You have checked that you are not creating a **duplicate** of an existing (maybe closed) request (use the search)
- [x] You have provided a short but meaningful title
- [x] You have provided enough information (such as links to web sites, blogs, technical description) so that anyone can understand what you are talking about. Just any only a title is not enough.
- [x] You have provided objective **facts**, not your personal feelings, your personal preferences.

#### Detailed description of the feature request:

_Please provide information (such as links to web sites, blogs, technical description)._

Currently, it is not possible to cancel e-step tuning (other than resetting the printer). Since the tuning takes some times, it is annoying. It is perhaps possible to implement canceling like I did for Auto PID Tuning.

andrivet, 2018-10-21:
Implemented in version 4.0.0 Alpha 1

galfisk, 2021-11-29:
I can't find this feature in 5.0.1, has it disappeared? After choosing the temperature for tuning, there is no button for cancelling or going back.

andrivet, 2021-12-24:
It is un unfortunate side-effect of the port to Marlin 2. I will fix this in a future release.

andrivet, 2022-07-11:
Related to #274 (for the technical solution). Implemented in version 5.4.0