andrivet/ADVi3pp #178: [Not an Issue] Just letting you know about the new EXTENSIBLE_UI API for Marlin

marcio-ao, 2018-10-15:
Hello,

I just recently became aware of the work you are doing and I really like the user interface you came up with! It looks really nice!

So I have been doing some work to standardize an API for Marlin to make it easy for people to write C++ code for custom touch user interfaces. The API at the moment consists of a few callbacks from Marlin, as well as several getter and setter methods for things that an UI may want to modify in Marlin.

If you ever get a chance, I would welcome any feedback on this API and whether this would be anything that would be useful to you. The hope is that by having a standard API, it would make it easier for developers to maintain their own UI code without having to deal with changes in upstream Marlin that would break their work.

The proposed API is called "EXTENSIBLE_UI" and the code [is here](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/Marlin/src/lcd/extensible_ui). To use it, a developer would need to do the following steps:

  - Enable "EXTENSIBLE_UI" in "Configuration.h"
  - Drop the UI source code in "src/lcd/extensible_ui/lib". The file "dummy.cpp" has examples of the callbacks functions that can be defined.
  - Use the methods defined in "src/lcd/extensible_ui/ui_api.h" to get and set internal Marlin state.
  - The API only provides an entry point and callbacks, the developer is responsible for all drawing operations and communications with their display module.

We are planning to using this API in our own open-source printers, but I would definitely welcome any feedback from other developers working on similar projects!

Thank you!

-- Marcio

andrivet, 2018-10-15:
Change the title to make it a issue (enhancement). Will look at it when porting ADVi3++ to Marlin 2.0

diver197, 2020-02-24:
Did you know that it is not possible to use dwin dgus display directly, and all data must be converted through OS once before UI display

andrivet, 2020-02-24:
@diver197 No idea of what you are talking about. Can you explain?

diver197, 2020-02-24:
Dwin dgus is composed of OS and UI. The open source part is only the UI and Marlin part. The OS part needs to be developed

andrivet, 2020-02-24:
There is no need to develop the OS. This printer is using Mini DGUS, not DGUS

andrivet, 2020-06-05:
@marcio-ao Just for information, I am currently porting ADVi3++ to Marlin 2 and yes, I am using as much as possible ExtUI. Some callbacks are missing for my port but I think they can be profitable for all, so I may send some PR to the Marlin upstream repository. I am waiting to have a stable version before doing that.

ShenRen, 2020-06-10:
That sounds great. I'm really looking forward to your project

andrivet, 2020-07-26:
Implemented in upcoming version 5

closed 2020-07-26 by andrivet