andrivet/ADVi3pp #180: Extruder and Bed Temp SetPoints Not Updated on Second Part Print From SD

Salty-Doug, 2018-12-02:
- Exact model of your printer: **Mono Maker Select Plus (Purchased Oct 2018)**
- ADVi3++ exact version: **3.0.2 Main and LCD**
- Printing using SD card or USB: **SD Card**
- Creating GCODE files with Cura v3.6.0.
- STL files are downloaded from Thingverse and created by Autodesk Fusion 360.

- After power up I select a file from the SD card and in print graph the temperature setpoints are correctly displayed. When the part is complete, I select a new file from the SD card the
 setpoints for the extruder and bed are at 0. If I go the Controls/Preheat menu and select temps they display a 0 in the graph. I have to press CoolDown then Preheat several times for it to take. On the third part I must power-cycle the unit.

- Expected behaviour: The temperature setpoints should be set from the next gcode file.
- Actual behaviour: They remain set to 0 after the cool down command from the previous part completion.
- I have run the this experiment several times today printing small parts for a personal project.
- I did not see this behavior with the "As Shipped" software.

Thanks in Advance.
Salty Doug

andrivet, 2022-03-25:
So far, I am not able to reproduce this issue. Do you still have it?

andrivet, 2022-05-21:
Still not able to reproduce, but maybe I do not understand well the different steps to reproduce it. Closing it for the moment, but will reopen if I get more information.

closed 2022-05-21 by andrivet