andrivet/ADVi3pp #182: Version 4.0.0 Alpha 2 known bugs

andrivet, 2018-12-10:
Thank you for submitting your feedback to the ADVi3++ project.

Please be sure to review the checklists below before submitting and use one of the templates. 

**It is mandatory**. If you do not comply with these rules, I will **delete** your issue.

- [ ] This issue is **NOT** about support. **This is not the right place to ask for support**. You **may** get support from the [ADVI3++ Facebook page](https://www.facebook.com/advi3pp/). You can get support from the developer of ADVi3++ from its [Patreon page](https://www.patreon.com/andrivet).

------------------------------------------------------------
### Feature Request

- [ ] Your request is not specific to your setup but can benefit to many people
- [ ] You have checked that you are not creating a **duplicate** of an existing (maybe closed) request (use the search)
- [ ] You have provided a short but meaningful title
- [ ] You have provided enough information (such as links to web sites, blogs, technical description) so that anyone can understand what you are talking about. Just any only a title is not enough.
- [ ] You have provided objective **facts**, not your personal feelings, your personal preferences.

Note: It is no more necessary to add "[FR]" in the title. You can, if you want, label your issue with "enhancement".

Detailed description of the feature request:

_Please provide information (such as links to web sites, blogs, technical description)._

------------------------------------------------------------
### Bug Report

- [ ] You have checked that you are not creating a **duplicate** of an existing (maybe closed) issue (use the search)
- [ ] You have tested with the **latest** version of ADVi3++ (including Beta releases)
- [ ] You are able to reproduce the bug
- [ ] You are able to describe step by step the process to reproduce the bug

* Exact model of your **printer**:
* **ADVi3++** exact version:
* Printing using **SD card** or **USB**:
* **For USB printing only** Software used to print with its exact version:

#### Description of the bug

* Expected behaviour:
* Actual behaviour:
* Steps to reproduce:
  - Do this
  - Do that

contraspec, 2018-12-13:
Di3 plus mk2 _ 2018
# as mentioned above, the print settings not working.
# PID bed tuning fails on temps above 80c.
# homing - homes ok, then parks the head on X100, Y50, Z1mm ( not at X0, Y0, Z10) etc. 
# Once print starts, Z offset from bed too high, after the bed level. Nozzel approx 1mm off the bed.
# possible bug, all axis ran away on stopping print, then starting again. GCODE has home all in start script.

churd83, 2018-12-14:
I think I have a bug that happens in When the print starts and runs through my Start Script. 

The problem is that when it goes to prime the nozzle it will sometimes Retract instead of Extrude. This seems to happen after using the touchscreen to feed or un-feed the nozzle. I'll do some more testing but it also happened in 3.0.2 as well.

This is my Start Script:
M140 S[first_layer_bed_temperature]
M109 S[first_layer_temperature] ;set hotend temp, no wait.
M116 ;waiting for both temperatures
G21 ; set units to millimeters
G90 ; use absolute positioning
M82 ; absolute extrusion mode
G28 ; home all axes
G29 ; ABL
m420 S1 ; Enable compensation using current grid/mesh
G1 X0 Y0 F5000
G1 Y5.0 F1000.0 ; go outside print area
G1 X60.0 Z0.2 E9.0 F1000.0 ; intro line
G1 X100.0 Z0.2 E21.5 F1000.0 ; intro line
G92 E0.0 ; reset extruder distance position

contraspec, 2018-12-14:
> Is your Mk2 stock?
only M/S hotend upgrade, few minor improvements, but all stock parameters (i.e positions and offsets)

Salty-Doug, 2018-12-14:
Hi Sebastian,
I have a Monoprice MSP that I purchased in Oct. ’18. I have only added the Z braces and an extruder blower housing.
I upgraded the LCD and the mainboard with this file "ADVi3pp-Mainboard-4.0.0-alpha.2.hex."
I like the menu layout in 4.0.
When I home, manually level, or print a part the speeds and accelerations of the motors are twice as fast as 3.0.2.
I tried changing the speeds, acceleration, and jerk in the Settitngs/Motor menu but this had no effect on the motions.

I then upgraded the mainboard with this file “ ADVi3pp-Mark2-Mainboard-4.0.0-alpha.2.hex.
When I homed all axes the Z stage ran past the microswitch and crashed into the bed.

Fair winds and calm seas.

Salty Doug Simms

KJ6FZW

Salty-Doug, 2018-12-14:
Hi Sebastian,
Thanks for all the work on this project.
In this statement, 

> 
>> I upgraded the LCD and the mainboard with this file "ADVi3pp-Mainboard-4.0.0-alpha.2.hex.”
I meant to say “ I upgraded the LCD to 4.0 and I used this file "ADVi3pp-Mainboard-4.0.0-alpha.2.hex to upgrade the mainboard.

Fair winds and calm seas.

Salty Doug Simms

KJ6FZW

MartynWilliams, 2018-12-15:
Hi,
I flashed the LCD and the mainboard as per the directions ..No Issues
The auto bed level sequence seems to have a bug where there is a discrepancy in the reported nozzle position on the z-height tuning screen (it seems to get out of step if you make use of the multiplier buttons)
Adjusting the Z-Height without using the multiplier seems to work flawlessly.

andrivet, 2018-12-18:
From Phillip Stamper on Patreon:

Hey Sebastian, I upgraded to the newest firmware and loving it so far but I ran into one issue. I do most of my printing from an SD card. If there is only one GCode on the card I can select it and the printer prints just fine but if I have multiple gcodes on the same SD code and try to select one to print from the LCD screen they are all merged on one line and when I select that only line it will print a random GCode. I hope that makes sense?

crashtest22, 2018-12-20:
Monoprice Maker Select Plus
all metal micro swiss hot end, Z-braces, BlTouch, rear mount fan, and various other updates.
4.0.0 Alpha 2
USB latest octoprint

Was trying to start a print and the extruder temp was fluctuating from 235-245 like it was trying to do a pid tune.
Tried to use the PID tune in octoprint and it looked as though it was successful.
Attempted to start same print and had the same issues.

when you do the pid tune in octo the screen on the printer would go to the same screen as if you did the tune on the touchscreen and you could select the check mark for done.

Ran the tune on the touch screen and started print again.

so it seems it may be a issue with the octo not speaking to the mainboard properly.

Also in the pid tune on the printer it has 5 levels of tune? I assume
I tried to tune at 200C and 240C and select different profiles but all of them say 200C

I had also tried to print from the SD card and had the same results until I did the pid tune on the printer.

temps over the last hour have been 239.5-240.8 very steady.

andrivet, 2018-12-20:
@contraspec 

* PID bed tuning fails on temps above 80c.
-> I try at 85c and see nothing special. What do you mean by "fails"?

* homing - homes ok, then parks the head on X100, Y50, Z1mm ( not at X0, Y0, Z10) etc.
-> Yes, this is on purpose

* Once print starts, Z offset from bed too high, after the bed level. Nozzel approx 1mm off the bed.
-> I am not able to reproduce

* possible bug, all axis ran away on stopping print, then starting again. GCODE has home all in start script.
-> Not sure to understand

andrivet, 2018-12-20:
@Salty-Doug 

* When I home, manually level, or print a part the speeds and accelerations of the motors are twice as fast as 3.0.2.
-> For home, manually level, it is the expected behavior. For print, can you give more information because I can't reproduce.

* When I homed all axes the Z stage ran past the microswitch and crashed into the bed
-> You choose the wrong file. The file you flash expects a sensor instead of a z endstop.

andrivet, 2018-12-20:
@MartynWilliams 

> The auto bed level sequence seems to have a bug where there is a discrepancy between the reported nozzle position on the z-height tuning screen (it seems to get out of step if you make use of the multiplier buttons) one reports 1.0mm the other .8 or .9 mm

Sorry I do not understand. You says "between the reported nozzle position on the z-height tuning screen. Between the z-height tuning screen and what?

contraspec, 2018-12-20:
> 
> 
> @contraspec
> 
>     * PID bed tuning fails on temps above 80c.
>       -> I try at 85c and see nothing special. What do you mean by "fails"?
*  any setting above 80c i get "PID tuning failed" on the screen.

> 
>     * homing - homes ok, then parks the head on X100, Y50, Z1mm ( not at X0, Y0, Z10) etc.
>       -> Yes, this is on purpose
*  how can i get the home to home to X0,Y0,Z10 as its an issue for me. 

> 
>     * Once print starts, Z offset from bed too high, after the bed level. Nozzel approx 1mm off the bed.
>       -> I am not able to reproduce
*  Nozzel is not at the correct height off the bed, I.E. 0.2mm and not 1.0mm which is produced at prersent.

>     * possible bug, all axis ran away on stopping print, then starting again. GCODE has home all in start script.
>       -> Not sure to understand
*  this is intermittent, one print starts on correct position, stop the print after seeing the print height not correct, then clean the bed and re level using abl, click start print as it has the home command in the start script, it homes but then this time does not start in the correct position, but then starts in a random location z40, y1000, x50 was approx start position.  printing off the sd card, card was not removed from printer during this time

MartynWilliams, 2018-12-21:
> @MartynWilliams
>
>
> Sorry I do not understand. You says "between the reported nozzle position on the z-height tuning screen. Between the z-height tuning screen and what?

The Z height reported between the + and - signs gets out of step with the Z Height in the section where the temperatures and the BLtouch icon is. In most cases they are the same , but occasionally they report different values. 
[image]
[image]

contraspec, 2018-12-23:
How do you expect me to do that? Whilst i am supporting this project for a working final firmware. I am not a programmer nor have the experience to compile on a firmware for my printer of which Sebastien is the first to develop a close to functional 3rd party firmware. If you could provide more info on how to, i could try it. But i do feel the current home location is not normal

andrivet, 2018-12-23:
@contraspec This homing on the center is made on purpose. This is not a bug. Please do not use this repository on GitHub for anything except bug reporting or feature request. Support is exclusively on Patreon.

crashtest22, 2018-12-24:
possible bug...not sure how to reproduce.
e steps changed itself to 165 plus
took me 2 small prints to figure out I was WAY over extruding.
my printer is now set to 90 something and working fine...
if it jumps itself again I will report

andrivet, 2018-12-29:
Closed in favor of #186

closed 2018-12-29 by andrivet