andrivet/ADVi3pp #184: BlTouch reports "Error: Probing Failed" instead of leveling

captroper, 2018-12-20:
Thank you for submitting your feedback to the ADVi3++ project.

Please be sure to review the checklists below before submitting and use one of the templates. 

**It is mandatory**. If you do not comply with these rules, I will **delete** your issue.

- [ X] This issue is **NOT** about support. **This is not the right place to ask for support**. You **may** get support from the [ADVI3++ Facebook page](https://www.facebook.com/advi3pp/). You can get support from the developer of ADVi3++ from its [Patreon page](https://www.patreon.com/andrivet).

------------------------------------------------------------
### Feature Request

- [X ] Your request is not specific to your setup but can benefit to many people
- [X ] You have checked that you are not creating a **duplicate** of an existing (maybe closed) request (use the search)
- [X ] You have provided a short but meaningful title
- [X ] You have provided enough information (such as links to web sites, blogs, technical description) so that anyone can understand what you are talking about. Just any only a title is not enough.
- [X ] You have provided objective **facts**, not your personal feelings, your personal preferences.

### Bug Report

- [X ] You have checked that you are not creating a **duplicate** of an existing (maybe closed) issue (use the search)
- [ X] You have tested with the **latest** version of ADVi3++ (including Beta releases)
- [ X] You are able to reproduce the bug
- [ X] You are able to describe step by step the process to reproduce the bug

* Exact model of your **printer**:
* **ADVi3++** exact version:
* Printing using **SD card** or **USB**:
* **For USB printing only** Software used to print with its exact version:

#### Description of the bug

(Unclear about the distinction between support and bug reporting. I figured you can delete it if you want to) 

* Expected behaviour:
BlTouch proceeds with leveling using touchscreen or G29 command  

* Actual behaviour:

BlTouch goes to level at the first point, touches the bed with the deployed probe, retracts the probe, raises the nozzle and displays a leveling error on the touchscreen, or "Error: Probing failed" when G29 sent via octoprint console. 

* Steps to reproduce:
 
1) Modified the following firmware items prior to building to correspond with where the BlTouch is mounted: 

define X_PROBE_OFFSET_FROM_EXTRUDER +45 // X offset: -left +right [of the nozzle]

DEFINE Y_PROBE_OFFSET_FROM_EXTRUDER -54 // Y OFFSET: -FRONT +BEHIND [THE NOZZLE]

// Set the boundaries for probing (where the probe can reach). 
#define LEFT_PROBE_BED_POSITION 45 
#define RIGHT_PROBE_BED_POSITION 195 
#define FRONT_PROBE_BED_POSITION 0 
#define BACK_PROBE_BED_POSITION 141

2) Ran Z-height via firmware

3) Ran leveling via firmware / and / or G29 via octoprint. 

Additional relevant? Information. 

Printer: Maker Select Plus 
Firmware version: 3.02 (modified only with the above changes)

Also tried the testing code found here: 

1) Blue LED will be ON if a servo signal is present. Blue LED is faint.  
Result: Blue LED on  
2) M119 ; should return OPEN. If not check that the BLTouch is enabled in the configuration. If enabled then check wiring and check that the correct pin is being used.  
Result: Open  
3) M280 P0 S10 ; should deploy the probe and the orange LED will be OFF.  
Result: probe acted as it should  
4) M119 ; should return OPEN  
Result: Open  
5) M280 P0 S90 ; should stow the probe and the orange LED will be ON  
Result: probe acted as it should  
6) M280 P0 S60 ; puts it into the M119 test mode.  
The probe should remain stowed and the blue LED should be OFF.  
Result: probe acted as it should  
7) M119 ; should return TRIGGERED  
Result: **Probe returned OPEN**  

As stated, in 7) the probe returned "Open" instead of Triggered as it presumably should have by being in test mode.

andrivet, 2018-12-20:
What you are asking is **support** for your custom build. I do not provide support here, as it is mentioned. Do not tick all the checkboxes without reading them.

closed 2018-12-20 by andrivet

captroper, 2018-12-20:
Understood, I actually just signed up for the patreon intending to post it there just in case.