andrivet/ADVi3pp #186: Version 4.0.0 Alpha 3 know issues

andrivet, 2018-12-29:
This issue is an inventory of all known bugs of version 4.0.0 Alpha 3. Checked items are fixed in Alpha 4.

* [ ] Runout sensor is not implemented
* [ ] When a PID tuning is finished, if you press back, you are back to the Temperature Graph page, not the PID tuning page. 
* [ ] When a PID tuning is finished, the temperature displayed is the default one, not the chosen one
* [ ] Sometimes when hitting Pause, the head goes in the corner as expected but then suddenly goes over the piece being printed.

andrivet, 2019-01-04:
@contraspec 
* "z height offset adjusted as close as possible to 0.1mm as per printer settings". Can you elaborate a little? I do not understand. Have you used the z offset tuning?
* "auto bed level only measures 70% of bed , alpha2 used whole surface". Have you set the right sensor in the sensor menu?

contraspec, 2019-01-04:
* Z height, adjusted in tunables, using a feeler guage for the nozzel gap between bed. 
* i used the default setting, as the setting was/looked correct, with the position.

andrivet, 2019-01-04:
@contraspec 
* What is the procedure to tune the z-height with the stock firmware?
* I do not think the default sensor (ADVi3++ Left Side) is correct. The sensor of the Mark II is at a very different position if I am not wrong.

contraspec, 2019-01-04:
On the stock, z height is set during the bed level procedure,

Press bed level, then head moves to bed center, adjust the offset with 0.1mm, then press next. Once pressed next it proceeds to do the auto bed level.

The sensor on my unit is dead centre to the nozzel, and 60mm to the rear.   Blue, with the following number   LJC12A3-A-Z/BY
PNP with 5mm spacing ?

andrivet, 2019-02-16:
Close in favor of #195

closed 2019-02-16 by andrivet

tobashadow, 2019-02-20:
Feature request, 

add 0.01 to values on baby stepping and sensor z-height tuning, 0.04 as the lowest adjustment is not fine enough adjustment for some surfaces.

andrivet, 2019-02-20:
@tobashadow this issue is closed. Please do not post under it. For feature request, you have to create an issue.