andrivet/ADVi3pp #189: #Feature Request. LED support in the LCD.

cobramike, 2019-01-03:
### Feature Request

LED support in the LCD.

So i got it to work. Problem is now that there is no LCD control for it . But marlin code is working. I use a WS2812B Strip. 

https://nl.aliexpress.com/item/DC5V-1m-4m-5m-WS2812B-Smart-led-pixel-strip-Black-White-PCB-30-60-144-leds/32337440906.html?spm=a2g0s.9042311.0.0.1ec64c4dhhAtWO

And i use on the wanhao plus pin 32. And its working nice And look verry nice. If you want i can make a smal tutorial. I must say after done this i can see the prints in the dark. And i now why the .... do you want LED. Its 2019 and i mis that. Also i am working on a runout filament sensor

andrivet, 2019-01-03:
How many LEDs do you have. If I read well, the power consumption is high (9 watts for 30 LEDs). Where do you take the power?

cobramike, 2019-01-04:
Sorry just finnist my work. Ill post my results. But i have 62 led and a LM2596 ( https://www.banggood.com/LM2596-DC-DC-Verstellbar-Step-Down-Schaltregler-Power-Supply-Module-p-88252.html?rmmds=search&cur_warehouse=CN ) Also i use this case inside the bothem case whare the power suply is. ( https://www.thingiverse.com/thing:1625233 )

cobramike, 2019-01-04:
And here are my pics. 

``
[image]
[image]
[image]

Kind regards
Mike

andrivet, 2019-01-04:
@cobramike It is clear now. Thanks for the info and the picture.

cobramike, 2019-01-04:
Your welcome.

rahulvij, 2019-05-16:
@cobramike I intend to add filament runout sensor, Do you mind sharing the wiring photo? I already have soldered pins (for BLtouch). All of them are free except 1 (i.e. 40). Since GND is already used, how did you wire them? 
Second is - what changes you made to the Marlin code, I assume you added **FIL_RUNOUT_PIN** in the **_pins_I3_PLUS.h_**, can you please confirm.
Thanks