andrivet/ADVi3pp #193: Unloading filament does not work

vaindil, 2019-01-13:
- [x] This issue is **NOT** about support.

------------------------------------------------------------
### Bug Report

- [x] You have checked that you are not creating a **duplicate** of an existing (maybe closed) issue (use the search)
- [x] You have tested with the **latest** version of ADVi3++ (including Beta releases)
- [x] You are able to reproduce the bug
- [x] You are able to describe step by step the process to reproduce the bug

* Exact model of your **printer**: Monoprice Maker Select Plus
* **ADVi3++** exact version: 3.0.2
* Printing using **SD card** or **USB**: OctoPrint (USB)
* **For USB printing only** Software used to print with its exact version: OctoPrint 1.3.10

#### Description of the bug

* Expected behaviour: Using the unload function should unload filament
* Actual behaviour: Unload function does nothing
* Steps to reproduce:
  - Attempt to unload filament

This has never worked since I installed v3.0.2, but it has never been an issue until now. I've always just cut off my filament and pushed the new one through. I need to change my nozzle, however, so I need to actually unload now. I tap the unload button and nothing happens. The nozzle is at the target temp, but the printer makes no noise and the filament doesn't move. Loading filament works perfectly, but unloading has never worked.

andrivet, 2019-01-13:
I am not able to reproduce this behavior and nobody except you reported this. So it is strange.

Prog72, 2019-01-13:
I always need to gently pull the filament out but isn't that normal?

andrivet, 2019-01-13:
@Prog72 Yes. But what @vaindil says (if I understand well) is that the E motors does not move at all. And this is not normal.

vaindil, 2019-01-13:
I pull on the filament incredibly hard, it doesn't budge at all. The printer doesn't make a noise like it used to when unloading would start.

Is it possible this is somehow a hardware issue with my printer specifically? I'm not asking for support with that, but internally it seems from a hardware perspective that unloading should work if loading does.

andrivet, 2019-01-13:
@vaindil Load and Unload code are very similar. The only difference is a "G1 E-1 F120" instead of "G1 E1 F120". So, unless someone else reports the same behavior, I would guess it is specific to your printer. For the moment, I leave this issue open.

wmarchewka, 2019-01-26:
This is happening to me now.  I unloaded, waited until temp and it pushed the filament out.  I then went back to the filament screen, and went into load.  And the "Wait until filament comes out" was displayed.  A power down of the A/C as well as the USB did not fix it.  It is stuck.

closed 2019-05-26 by andrivet