andrivet/ADVi3pp #194: Z Print Height Not Correct

Salty-Doug, 2019-01-15:
Thank you @andrivet , for your work on this control software.

### Bug Report

- [ X ] You have checked that you are not creating a **duplicate** of an existing (maybe closed) issue (use the search)
- [ X ] You have tested with the **latest** version of ADVi3++ (including Beta releases)
- [ X ] You are able to reproduce the bug
- [ X ] You are able to describe step by step the process to reproduce the bug

* Exact model of your **printer**: Monoprice MSP Mark 1
* **ADVi3++** exact version: 4.0.0 a3
* Printing using **SD card** or **USB**: SD Card
* **For USB printing only** Software used to print with its exact version:
Additions:
Z-Tower Braces.
3mm glass bed.
New BLTouch Smart. (Connected using your excellent directions.)
I added an Autolevel, G29, to the Start G-Code.

#### Description of the bug

* Expected behaviour:
After calibrating the BLTouch Z-Height with a 0.1mm feeler gage  and performing an Autolevel function. I expect the nozzle to bed gap to be close to 0.1mm during printing.

* Actual behaviour:
After calibrating the BLTouch Z-Height with a 0.1mm feeler gage  and performing an Autolevel function. I observed the nozzle to bed gap to be 0.7mm during printing.
I also went to the Sensor Settings screen and manually adjusted the Z-Height. then re-printed the sample and observed little change.

* Steps to reproduce:

 - I removed the filament from the print head and powered down the printer.

 - After power-up, home the printer using the Controls/Move menu, and pre-heat the nozzle and bed, then select the file to print.

 - I wait for the nozzle to start printing the outer lines of the part then turn off the power switch to freeze the stage. 
 - Then I measure the gap between the nozzle and bed with a feeler gage. The gap was 0.7mm on the first print.
 - I changed the Z-Height from -2.55 to -3.25 and on the next print, the gap had changed to 0.6mm.

andrivet, 2019-01-15:
@Salty-Doug I keep this issue for the moment, but in the futur, please do not open issues for alpha or beta releases.

Salty-Doug, 2019-01-15:
Sure, How should I communicate what I find on the alpha release?

andrivet, 2019-02-16:
@sakunamary There is an issue dedicated for that: https://github.com/andrivet/ADVi3pp-Marlin/issues/186

closed 2019-05-26 by andrivet