andrivet/ADVi3pp #195: Version 4.0.0 Beta 1 known issues

andrivet, 2019-02-16:
This issue is an inventory of all known issues of version 4.0.0 Beta 1. Checked items are fixed in Beta 2 (or final).

* [ ] Sometimes when hitting Pause, the head goes in the corner as expected but then suddenly goes over the piece being printed.
* [ ] (Sometimes) when pressing Back in the Load/Unload screen, the E stepper motor does not stop.
* [ ] The Linear Advance screens are still under Tuning but they will be removed in this version.

MartynWilliams, 2019-02-17:
Hi,
I noticed that the Z-Height tuning has adjustment values which are different from the previous releases (.04, .12 and 1.0) and the adjustments work as they should. There is however a strange artefact, when you press the buttons, the values V1, V2 and V3 appear  (leave your finger on the button and peek)

contraspec, 2019-02-20:
Hi,

Possible issue, not sure if it is, but works fine on stock firmware.
Once a print has started and you need to cancel, by pressing stop on LCD, the following happens
* ADVi3++4.0.0_B1 - printer stops printing, and nozzle doesn't lift off the print, but stays in situ melting the printed part, cant move the head by hand as the motors are still locked, now need to shuffle through the menu to either move axis or disable motors. either way printed part ruined.
* stock firmware - once stop is triggered, nozzle raises enough to clear the part, then homes to X & Y.

on a different note, the printer is running really stable, 14hr print successful. smooth and quiet compared to before. well done on this great piece of firmware.

Di3+ mk2
4.0.0_Beta1  Build_190216163648

andrivet, 2019-02-21:
@contraspec ADVi3++ follows normal Marlin behavior, so it is not really an "issue", more a different behavior. But I agree that it is not optimum. What you can do is to press "Pause" instead of "Stop". Are you printing from USB or SD?

flyingelephant19, 2019-02-22:
Hi

using cocoon create touch model (wanhao i3 plus) i have a problem with what seems to be speed? 

it seems X axis in particular jitters very aggressive and cannot print as a result. to test this, i switched back to ADV 3.0.2 and the problem has disappeared. 
what i have noticed is the 3.0.2 the speed at which the head moves and all axis move is smoother and slower, whereas beta 4.0 moves very aggressive in comparision.

thanks. i hope this was helpful.

flyingelephant19, 2019-02-22:
i think i found the problem!

for some reason the ACCELERATION value gets jumbled up or rather it is not set right. so bascially ACCELERATION was set as follows somehow.

max x = 195
max y = 60
max z = 0
max e = 25
print = 0
retr = 0
travel = 0

i can confirm that this is the problem. 

could you check to see if somehow these settings are getting reset or are defaults?

i had to set new values and m500 them. the values i took from ADV 3.0.2.

andrivet, 2019-02-22:
@flyingelephant19 These are not the defaults. Do you upgrade from 3.0.2 directly to 4.0.0? It is supposed to display a warning and reset the defaults just after the upgrade.

flyingelephant19, 2019-02-23:
yes i know they are not the defaults but for some reason the machine entered them in on its own. no jokes. i never had these settings in there prior to the upgrade which was from 3.0.2.

thecomputerguythatdoesntsuck, 2019-02-24:
Just for the sake of me and probably others being forgetful, can you add on the z-height screen what size of feeler gauge to use? I'm assuming it's 0.1mm but I could be wrong.

andrivet, 2019-02-24:
@thecomputerguythatdoesntsuck This has nothing to do with the firmware, up to you to decide. It depends of you bed material, your filament, the humidity, etc.

contraspec, 2019-02-24:
Hi Sebastien,

Printing with the followin :

from sdcard
v4_beta1
The following happens, in this sequence.

start print, (all starts fine, )
press pause as per previous msg to use pause instead of stop, head lifts up, then into far left corner.
press the pause/resume, and extruder purges aprox 100mm of filament, then resumes to print position.
Now this is where a possible bug shows up.
press the stop to cancel. It stops.
but now the lcd freezes up, touch the lcd buttons, it highlights and nothing happens, no beep/audio.
must push reset button or turn printer off to recover command of printer.

Salty-Doug, 2019-03-17:
Hi @andrivet 
I really like the 3 speed number changer system that you implemented for the Z-Height calibration.
There are several more places in Settings that could us this 3 speed number changer:

1. Sensor location, X, Y, & Z.
2. PID Parameters.
3. Motors, Steps, Feedrate, Acceleration & Jerk.

Thanks and keep up the good work.
Salty Doug

Salty-Doug, 2019-03-17:
Hi @andrivet 
Have you been able to duplicate #194 ?
I am seeing it again with 4.0.0 Beta 1.
I did not experience it at first but after I ran the PID Tuning of the extruder and the print bed I had problems with the filament not sticking to my glass plate. I have been using Aquanet spray successfully. I manually leveled the plate with a 0.06mm feeler gage and checked the nozzle height at printing and measured .55mm.
[image]

I retrograded back to 3.0.2 and the same part printed perfectly.

Thanks.
Salty_Doug

andrivet, 2019-03-20:
@Salty-Doug No, I was not. It may be related to another bug I am currently chasing.

Salty-Doug, 2019-03-20:
Ok @andrivet, Thanks! 
I like the Ver 4 features a lot. 

Fair winds and calm seas.

Salty Doug Simms

KJ6FZW

cyberhiker1965, 2019-03-25:
During the BED PID tune my results came out as:
`
Kp: 1125.28 Ki: 218.04 Kd: 1451.89

Recv: PID Autotune finished! Put the last Kp, Ki and Kd constants from below into Configuration.h

Recv: #define DEFAULT_bedKp 1125.28

Recv: #define DEFAULT_bedKi 218.04

Recv: #define DEFAULT_bedKd 1451.89
`

These seem to be off what I'd expect from a normal PID tune.  Related to this the buttons to adjust the tune numbers before saving do no allow these values to be entered (I guess because they are outside of the normal range?  When hitting + on Kp the number reset to zero. On Ki the number appeared to reset to 200 and would not allow adjustment past that value.

Spconnol, 2019-03-25:
When using the hotend fan in the preheat menu, it uses the S0-255 instead of a percentage. So if you set it to "100%" the fan's only blowing at 39%.

Salty-Doug, 2019-03-26:
After calibrating the Bed PID on my Monoprice MSP I found that the bed heats much quicker than V3.
After the calibration was complete I pressed the arrow in the upper right corner. I confirmed that the new values were entered into the Bed settings.

Best
Salty Doug

Spconnol, 2019-03-26:
> When using the hotend fan in the preheat menu, it uses the S0-255 instead of a percentage. So if you set it to "100%" the fan's only blowing at 39%.

Clicking cooldown also doesn't change the fan speed back to 0 if you use it in a preset preheat.

andrivet, 2019-03-31:
For BLTouch Smart V3 support:
* https://github.com/MarlinFirmware/Marlin/issues/13345
* https://github.com/MarlinFirmware/Marlin/pull/13406
* https://github.com/MarlinFirmware/Marlin/pull/13454
* https://github.com/MarlinFirmware/Marlin/pull/13495

andrivet, 2019-04-20:
Closing in favor of Beta 2

closed 2019-04-20 by andrivet