andrivet/ADVi3pp #200: Extruder Fan Off at Idle/"Room Temperature" (Noise Reduction Mode)

Walletclan, 2019-02-23:
It would be a nice feature to be able to have the extruder fan off when the machine is on and idle at room temperature. This could act as a noise reduction feature.

Maybe make it possible for the user to set the "room temp" for the extruder and if the extruder temp is <"Room Temp" the fan is off until given a command to print or move. This would also not engage during a cool down command.

Prog72, 2019-02-23:
I don't think there's any way to control the extruder fan. It's on whenever the printer's powered up.

andrivet, 2019-02-24:
Please:
* Use the template. This is mandatory.
* The extruder fan is always on because it is always connected to power.

closed 2019-02-24 by andrivet