andrivet/ADVi3pp #206: Version 4.0.0 Beta 4 known issues

andrivet, 2019-05-26:
This issue is an inventory of all known issues of version 4.0.0 Beta 4. Checked items are fixed in Final.

cjackson234, 2019-05-27:
BLTouch 3 Auto Leveling starts outside of the print bed on the front left corner. Auto Leveling then fails.

andrivet, 2019-05-27:
@cjackson234 have you selected the right sensor (position)?

cjackson234, 2019-05-27:
Moving the conversation there.

csammis, 2019-05-27:
- Maker Select Plus Mark 1 board
- Firmware version 4.0.0-BLtouch3-beta4
- BLtouch v3, manufacture date 2019-03-28
- Printer controlled via Octoprint 1.3.11, no plugins except Firmware Updater

When autoleveling at the start of a print with G29, the head will sometimes probe the front three points in midair. The BLtouch probe will deploy and the head will dip as though pressing to the bed but the head will raise again without the probe making contact with the bed. The second and third rows are leveled correctly in that the probe makes contact with the bed as expected.

This seems to be loosely correlated to changing the Z-height between prints. I'm still trying to tune my BLTouch setup and so I am starting and cancelling prints frequently. If I change the Z-height via Settings -> Sensor between prints, the next G29 will very likely have this issue. Happened every time (after I started keeping track) with 4.0.0-beta3 and has happened a couple of times today after updating to 4.0.0-beta4.

andrivet, 2019-05-27:
@csammis this is a problem with your setup and not with the firmware. No support here. Please post on Patreon

closed 2019-06-08 by andrivet