andrivet/ADVi3pp #207: May need 2-3 decimal places for Linear Advance

Mjankor, 2019-06-09:
On my machine I estimate a linear advance value of 0.065. I'm enabling it through Cura start Gcode at the moment, but it may also be nice to adjust it in the machine. Version 4.0 allows adjustment of linear advance through the menu, but only to 1 decimal place.

Cocoon Create Touch
ADVi3pp-Mainboard-4.0.0.hex
SD Card
0.6mm nozzle
ABS @ 250ºC

andrivet, 2019-06-09:
Please use the template for issues. This is **mandatory**. I delete issues that do not follow the rules.

For this particular point, your K value is very strange: K is around 0.1-2.0 (see http://marlinfw.org/docs/features/lin_advance.html).

Mjankor, 2019-06-09:
Sorry. I'll try and follow the template better.

Yes, it's a very low K-value, though I don't think it's unique. I've seen occasional other low values. e.g. 0.08 here:
https://www.reddit.com/r/3Dprinting/comments/9oo046/marlin_119_linear_advance_k_values/.

I'm guessing it's because I'm running a 0.6mm nozzle, and there isn't as much opportunity for pressure to build up.

I also ended up with two decimal places on my bowden kossel, with a value of 0.65. 2 decimals would be fine. From looking closely at the LA test pattern I doubt I'd be able to spot any difference between K values of 0.06 and 0.07.

andrivet, 2020-07-26:
Implemented in upcoming version 5.

closed 2020-07-26 by andrivet