andrivet/ADVi3pp #208: Homing Z moves extruder to X10, Y10

Prog72, 2019-06-09:
Thank you for submitting your feedback to the ADVi3++ project.

Please be sure to review our [code of conduct](code_of_conduct.md) and the checklists below before submitting and use one of the templates. 

**It is mandatory**. If you do not comply with these rules, I will **delete** your issue or comment. **I will be very strict**. If you persist, I may **ban** you.

- [x] This issue (or comment) is **NOT** about support. **This is not the right place to ask for support**. 
Instead, go to [ADVI3++ Community - Get Help](https://community.advi3pp.com/c/get-help). 

------------------------------------------------------------
### Feature Request

- [x] Your request is not specific to your setup but can benefit to many people
- [x] You have checked that you are not creating a **duplicate** of an existing (maybe closed) request (use the search)
- [x] You have provided a short but meaningful title
- [x] You have provided enough information (such as links to web sites, blogs, technical description) so that anyone can understand what you are talking about. Just any only a title is not enough.
- [x] You have provided objective **facts**, not your personal feelings, your personal preferences.

#### Detailed description of the feature request:

_Please provide information (such as links to web sites, blogs, technical description)._

------------------------------------------------------------
### Bug Report

- [x] You have checked that you are not creating a **duplicate** of an existing (maybe closed) issue (use the search)
- [x] You have tested with the **latest** stable version of ADVi3++
- [x] You are able to reproduce the bug
- [x] You are able to describe step by step the process to reproduce the bug

#### Description of the bug

* Exact model of your **printer**: Balco 3D Printer Touch (v5.1 mainboard)
* **ADVi3++** exact version (Mainboard): ADVi3pp-Mainboard-4.0.0.hex
* Printing using **SD card** or **USB**: USB
* **For USB printing only**: Software used to print with its exact version: Cura 4.0 via OctoPrint 1.3.11
* Expected behaviour: Homing Z axis shouldn't move X or Y axes
* Actual behaviour: Extruder moves to X10, Y10 (or thereabouts) whenever the Z axis is homed. Homing all axes moves the extruder to X10, Y10, Z0
* Steps to reproduce:
Either
  - Move extruder to any position
  - Home Z axis via controls on LCD or by sending G28 Z command.
or
  - Move extruder to any position
  - Home all axes via controls on LCD or by sending G28 command.

andrivet, 2019-06-10:
I am not able to reproduce what you see. You do not have a BLTouch, right?

Prog72, 2019-06-10:
That's right, I level manually.

chschmit, 2019-06-10:
This is exactly the issue I'm having. 
I do not have a BLTouch and I'm only seeing this behavior with v4.0.0; Z homes at X10Y10
All of the 3.0.x versions work for me where Z homes at X0Y0

andrivet, 2019-06-11:
Two people, same config, same issue => I probably have a bug here or a non-obvious behavior. I will plug back the Z endstop and make some news tests. I will let you know the results.

andrivet, 2019-06-11:
I think that the "problem" is in fact a feature: safe homing.

@Prog72 and @chschmit : can you send me a private message on [ADVi3++ Community](https://community.advi3pp.com/) so I can send you an updated firmware (without Safe Homing)? If it solves the problem, I will publish a patched version.

andrivet, 2019-06-11:
Safe Homing will be disabled for non-BLTouch builds, as it is confusing. Probably in version 4.0.1 in some days.

closed 2019-06-11 by andrivet