andrivet/ADVi3pp #209: Prints are not centered

andrivet, 2019-06-10:
Thank you for submitting your feedback to the ADVi3++ project.

Please be sure to review our [code of conduct](code_of_conduct.md) and the checklists below before submitting and use one of the templates. 

**It is mandatory**. If you do not comply with these rules, I will **delete** your issue or comment. **I will be very strict**. If you persist, I may **ban** you.

- [x] This issue (or comment) is **NOT** about support. **This is not the right place to ask for support**. 
Instead, go to [ADVI3++ Community - Get Help](https://community.advi3pp.com/c/get-help). 

------------------------------------------------------------
### Bug Report

- [x] You have checked that you are not creating a **duplicate** of an existing (maybe closed) issue (use the search)
- [x] You have tested with the **latest** stable version of ADVi3++
- [x] You are able to reproduce the bug
- [x] You are able to describe step by step the process to reproduce the bug

#### Description of the bug

* Exact model of your **printer**: Monoprice Select Plus
* **ADVi3++** exact version (Mainboard): 4.0.0
* Printing using **SD card** or **USB**: USB
* **For USB printing only**: Software used to print with its exact version: Simplify3D 4.1.2
* Expected behaviour: Print a 200x200 square. Expect the print to be centered
* Actual behaviour: It is not centered in both X and Y directions
* Steps to reproduce:
  - Print a 200x200 square

chschmit, 2019-06-10:
I concur with this...

It appears that when my startup script does the initial homing sequence it zeros at X0,Y0,Z0 momentarily, but jumps to X10,Y5,Z0 before running the rest of my startup script. From there any programs with X locations beyond 190 or Y locations beyond 195 crash into the extents of the printer. This didn't happen with 3.x firmware.

andrivet, 2019-06-10:
After further tests, I see that my previous test was wrong. Prints are centered on the X-axis (+/- 0.5mm). On the Y-axis, there is a small offset (4mm).

chschmit, 2019-06-10:
Strange. I'm definitely off by 10mm in X and 4 or 5mm in Y.

I'm using the same starting script for both 3.x and 4.0 firmwares (I have two printers).

Prog72, 2019-06-10:
I wonder if this is linked to #208 where the extruder moves from X0, Y0 during homing.

andrivet, 2019-06-10:
@chschmit like i said, the behavior is different between version 3 and 4. This is intended.
@Prog72 please do not mix things randomly, it does not help.

Prog72, 2019-06-10:
Apologies.

andrivet, 2019-06-11:
Will be fixed in version 4.0.1, in some days.

andrivet, 2019-06-30:
Was in fact not fixed in 4.0.1. Will be in 4.0.2

closed 2019-06-30 by andrivet