andrivet/ADVi3pp #215: Sometimes after flashing, the LCD Panel displays a version mismatch warning

andrivet, 2019-07-10:
- [x] This issue (or comment) is **NOT** about support. **This is not the right place to ask for support**. 
Instead, go to [ADVI3++ Community - Get Help](https://community.advi3pp.com/c/get-help). 

------------------------------------------------------------
### Bug Report

- [x] You have checked that you are not creating a **duplicate** of an existing (maybe closed) issue (use the search)
- [x] You have tested with the **latest** stable version of ADVi3++
- [x] You are able to reproduce the bug
- [x] You are able to describe step by step the process to reproduce the bug

#### Description of the bug

* Exact model of your **printer**: Wanhao Duplicator i3 Plus
* **ADVi3++** exact version (Mainboard): 4.0.2
* Printing using **SD card** or **USB**: n/a
* **Do you use a BLTouch?**: no
* Expected behaviour: The firmware should detect the versions of the mainboard and of the LCD are the same
* Actual behaviour: The LCD Panel displays a version mismatch warning with LCD: 4.2, Mainboard: 4.0.2

andrivet, 2019-07-10:
The LCD version displayed on the page is the wrong one: it is the DGUS firmware version and not the ADVi3++ LCD version.

andrivet, 2019-07-11:
Fixed in 4.0.3

andrivet, 2019-07-30:
Reopen because it is not completely fixed. Apparently, the way I get the version of the LCD resources is not reliable enough. Sometimes, it returns 0. So I guess, I will remove this detection mechanism in a future release.

sparkiesweden, 2019-09-04:
I have two identical (hardware-wise) I3+ mk2 printers, one shows the missmatch error and the other don't, otherwise stuff seems to work, except that it's not possible to turn off the touch-beep on either of the printers .. a bit annoying but perhaps not related....

redtmpusr, 2019-09-24:
Same situation as @sparkiesweden . Turning on lcd beep turns tone into almost double beep. Turned off the normal beep is there. Maker Select Plus with 4.0.3 lcd and 4.0.3 bltouch v3 mainboard. A version mis match was shown to me on fresh flash. The os reports main board as 4.0.3 and lcd as 49.2.14

closed 2019-12-31 by andrivet