andrivet/ADVi3pp #217: Y axis jogs itself on first G0/G1 after power on

arpreq, 2019-10-01:
Thank you for submitting your feedback to the ADVi3++ project.

Please be sure to review our [code of conduct](code_of_conduct.md) and the checklists below before submitting and use one of the templates. 

**It is mandatory**. If you do not comply with these rules, I will **delete** your issue or comment. **I will be very strict**. If you persist, I may **ban** you.

- [x] This issue (or comment) is **NOT** about support. **This is not the right place to ask for support**. 
Instead, go to [ADVI3++ Community - Get Help](https://community.advi3pp.com/c/get-help). 

------------------------------------------------------------
### Bug Report

- [x] You have checked that you are not creating a **duplicate** of an existing (maybe closed) issue (use the search)
- [x] You have tested with the **latest** stable version of ADVi3++
- [x] You are able to reproduce the bug
- [x] You are able to describe step by step the process to reproduce the bug

#### Description of the bug

* Exact model of your **printer**: Monoprice Maker Select Plus V1
* **ADVi3++** exact version (Mainboard): 4.0.3
* Printing using **SD card** or **USB**: N/A
* **For USB printing only**: Software used to print with its exact version: Octoprint 1.3.11
* **Do you use a BLTouch?**: No
* Expected behaviour: Selected axis is jogged
* Actual behaviour: Selected axis is jogged **AND** Y axis gets jogged in positive direction
* Steps to reproduce:
  - Turn the printer on from a powered-off state
  - Go into "Controls" menu
  - Go into "Move" menu
  - Click the X (either direction), Z (either direction), or Y (negative/top button only) jog buttons
  - Observe the Y axis move in the positive direction in addition to the requested move
    + Even if the Y-negative jog button is pressed, Y will still jog in the positive direction on first press

Notes:
  - Works with Octoprint connected or not
  - Sending equivalent G91+G0/G1 commands through Octoprint terminal exhibits same behavior
  - The unwanted Y axis jog happens simultaneously and has the same feed rate as the requested jog
  - Issue only happens on first axis jog after machine is turned on
  - The home buttons do not seem to trigger the effect
  - Jogging the Y axis in positive direction (bottom button) first results in normal behavior
  - The extra move does not appear in the Octoprint terminal. Output when moving the X axis on the LCD touch screen:
`Recv: echo:enqueueing "G91"
Recv: echo:enqueueing "G1 X4 F1000"
Recv: echo:enqueueing "G90"`

andrivet, 2019-10-08:
I can’t reproduce this behavior and you are the only one to report it so far. So I assume it is a problem with your printer, not with the firmware.

arpreq, 2019-10-08:
Thanks for replying. After a quick search, I found multiple people that complain in this thread (Gareth, Thomas_Smith, and Aaron) : https://community.advi3pp.com/t/x-and-y-axis-some-times-moves-on-their-own/175/6

I downgraded to 3.0.0 beta3 (which is what I upgraded to 4.0.3 from) and the issue disappeared. Reflashing 4.0.3 brings the issue back.

The behavior really seems to indicate that it might be something in the firmware with how repeatable this is. And the movement doesn't look like it's electronic noise-based, it looks (and feels) like an actual movement command is added in.

andrivet, 2019-10-09:
@arpreq The thread you mentioned is a mix of different problems. Some are however similar to what you see. What you describe is however quite reproducible. It is not the case for other people.
I problem I see in the past similar to what is described in the thread was on the X-axis. It moves alone sometimes. After research and help of some other people, I found it was electronic noise: it triggers the X endstop because of that. The solution was to disable the motors. It was possible in this very particular case.

For Z axis, I am not sure from your description, but I think it is a feature of the firmware to avoid colliding when homing x or y.

For Y axis my guess is that it is a combination of electronic noise and some changes in the firmware between versions 3 and 4. But in fact, I am not sure to fully understand what you described. For example the term "jog" is not clear for me. I am also not sure if your Y axis is moving only 1 step, or always.

Another question: Is it really annoying (if it moves only 1 step)?

arpreq, 2019-10-09:
> But in fact, I am not sure to fully understand what you described. For example the term "jog" is not clear for me. I am also not sure if your Y axis is moving only 1 step, or always.

Sorry for the confusion - I used the terms "move" and "jog" interchangeably. The Y axis does a full move, as if a G0/G1 command was given that included it.
For example, right after powering on if I send the printer `G0 X50 F1000`, the printer will act exactly as if I sent `G0 X50 Y50 F1000`. The Y axis is even left powered on, so the bed can't be moved by hand unless I hit the "motors off" button.

Here is a small video showing the phenomenon: https://www.dropbox.com/s/1psap8r6f7evh3u/20191009_172555_1.mp4?dl=0

> For Z axis, I am not sure from your description, but I think it is a feature of the firmware to avoid colliding when homing x or y.

You're probably right - the auto Z axis movement might be intended to prevent the nozzle from scraping across the bed as it homes. I'm not super worried about this, but just wanted to mention it because it seemed related.

> Another question: Is it really annoying (if it moves only 1 step)?

Since the Y axis is copying whatever the first move is, it's a little dangerous in my opinion. If the Y axis does not have enough room for the move, the bed will crash into the frame.

contraspec, 2019-10-10:
Hi Sebastien,

im having a similar situation, where if i jog the Z-axis upwards to lift the hot-end, the Y-axis jogs a few mm and stops whilst the Z-axis continues to jog. very strange behavior, which should not happen.

i do believe this could be related to the op's post. 

FW_4.0.3
Di3+mk2
no g-code commands given, no sdcard or usb, only freshly powered on, and jog to lift z axis to clean nozzle.

andrivet, 2019-10-10:
@contraspec what do you mean by
> i do believe this could be related to the **op's post**

Aianir, 2019-11-02:
I have the exact same issue as described in this bug report. If I follow the steps to reproduce, the printer performs the same unwanted Y-positive movement.

Exact model of your printer: Wanhao Duplicator i3 Plus
ADVi3++ exact version (Mainboard): 4.0.3
Do you use a BLTouch?: Yes (but not triggered when reproducing the unwanted behaviour)

I also have another issue that I am unable to reproduce; twice after printing, after homing X and Y and is supposed to move the bed to Y200, it moves the X and Y at an incredible speed to about X100 Y100 (maybe the safe home coordinates). If I find any pattern in this I will write up a real bug report, just thought I'd mention it.

Salty-Doug, 2019-11-03:
Sebastian, 
I can reproduce this behavior repeatedly. If I Stop a print then try to home or move an axis, the stage will move the Y and Z axis until I power the printer off.

andrivet, 2019-11-12:
@Salty-Doug Are you describing the same behavior that in this issue or another one? Looks like it is another one: the issue steps does not talk about "stop a print". So please fill another issue.

andrivet, 2019-11-12:
@arpreq I am now able to reproduce this bug. Will look at it to fix it.

arpreq, 2019-11-13:
Great, let me know if you require any more information.

Salty-Doug, 2019-11-14:
@andrivet I entered this issue as a seperate item #219 
Thanks.

andrivet, 2020-01-16:
Fixed in 4.0.5

closed 2020-01-16 by andrivet