andrivet/ADVi3pp #218: Numbers for Teaching Tech Front BlTouch Mount Return "(L,R,F,B) out of bounds." error.

captroper, 2019-10-20:
Thank you for submitting your feedback to the ADVi3++ project.

Please be sure to review our [code of conduct](code_of_conduct.md) and the checklists below before submitting and use one of the templates. 

**It is mandatory**. If you do not comply with these rules, I will **delete** your issue or comment. **I will be very strict**. If you persist, I may **ban** you.

- [ X] This issue (or comment) is **NOT** about support. **This is not the right place to ask for support**. 
Instead, go to [ADVI3++ Community - Get Help](https://community.advi3pp.com/c/get-help). 

------------------------------------------------------------
### Feature Request

- [X ] Your request is not specific to your setup but can benefit to many people
- [ X] You have checked that you are not creating a **duplicate** of an existing (maybe closed) request (use the search)
- [ X] You have provided a short but meaningful title
- [ X] You have provided enough information (such as links to web sites, blogs, technical description) so that anyone can understand what you are talking about. Just any only a title is not enough.
- [ X] You have provided objective **facts**, not your personal feelings, your personal preferences.

#### Detailed description of the feature request:

_Please provide information (such as links to web sites, blogs, technical description)._

------------------------------------------------------------
### Bug Report

- [ X] You have checked that you are not creating a **duplicate** of an existing (maybe closed) issue (use the search)
- [ X] You have tested with the **latest** stable version of ADVi3++
- [ X] You are able to reproduce the bug
- [ X] You are able to describe step by step the process to reproduce the bug

#### Description of the bug

* Exact model of your **printer**: Monoprice Maker Select Plus
* **ADVi3++** exact version (Mainboard): ADVI3++ 4.03
* Printing using **SD card** or **USB**: NA
* **For USB printing only**: Software used to print with its exact version:
* **Do you use a BLTouch?**: If yes, exact version of the BLTouch: Yes, Smart v. 2.0 I believe
* Expected behaviour: Entering the values for the Teaching Tech Front BlTouch Mount (+10, -38) into either the custom slot, or changing the values of one of the other 2 to match those values properly adjusts boundaries to be probed. 
* Actual behaviour: When you try to level the bed with either G29 command or LCD leveling it does not work. LCD will continue saying 'homing', but not display anything else. Octoprint says " ? (L,R,F,B) out of bounds."
* Steps to reproduce:
1) On the LCD Screen click Settings > Sensor> 
2) Adjust the values to be +10.0 and -38.0 
3) Click the checkmark at the top to confirm
4) Click on Controls (or send a g28 command to the printer) 
5) Click on leveling (or send a g29 command to the printer) 

Through troubleshooting, it appears that the error happens with any value over 7 for the X offset.

TriggerMeister, 2019-11-21:
I just encountered the same phenomenon after a Marlin upgrade.

#define NOZZLE_TO_PROBE_OFFSET { 24, 1, -1.45 } 

Send: G29
Recv: ? (L,R,F,B) out of bounds.
Recv: ok

Error reported by printer
Your printer's firmware reported an error. Due to that OctoPrint will disconnect. Reported error: Too far from reference point

cgsaldana, 2019-12-08:
I had a similar issue when I installed a custom designed BLTouch sensor.

I traced the issue to a simple typo in the Marlin.h file on line 642.
IS:
return position_is_reachable(rx - ADVi3pp::x_probe_offset_from_extruder(), ry - ADVi3pp::x_probe_offset_from_extruder())

SHOULD BE:
return position_is_reachable(rx - ADVi3pp::x_probe_offset_from_extruder(), ry - ADVi3pp::y_probe_offset_from_extruder())

andrivet, 2019-12-08:
Took me 5 minutes to figure out the difference. Thank you so much to figure out this nasty bug! Will be part of the next release (scheduled for past month, but will be in December instead).

andrivet, 2019-12-08:
Fixed by merge #221

closed 2019-12-08 by andrivet