andrivet/ADVi3pp #219: After stopping a print the X, Y, and Z axis will all move when Z jog is pressed.

Salty-Doug, 2019-11-03:
Thank you for submitting your feedback to the ADVi3++ project.

Please be sure to review our [code of conduct](code_of_conduct.md) and the checklists below before submitting and use one of the templates. 

**It is mandatory**. If you do not comply with these rules, I will **delete** your issue or comment. **I will be very strict**. If you persist, I may **ban** you.

- [ ] This issue (or comment) is **NOT** about support. **This is not the right place to ask for support**. 
Instead, go to [ADVI3++ Community - Get Help](https://community.advi3pp.com/c/get-help). 

------------------------------------------------------------
### Feature Request

- [ ] Your request is not specific to your setup but can benefit to many people
- [ ] You have checked that you are not creating a **duplicate** of an existing (maybe closed) request (use the search)
- [ ] You have provided a short but meaningful title
- [ ] You have provided enough information (such as links to web sites, blogs, technical description) so that anyone can understand what you are talking about. Just any only a title is not enough.
- [ ] You have provided objective **facts**, not your personal feelings, your personal preferences.

#### Detailed description of the feature request:

_Please provide information (such as links to web sites, blogs, technical description)._

------------------------------------------------------------
### Bug Report

- [ X] You have checked that you are not creating a **duplicate** of an existing (maybe closed) issue (use the search)
- [ X] You have tested with the **latest** stable version of ADVi3++
- [ X] You are able to reproduce the bug
- [ X] You are able to describe step by step the process to reproduce the bug

#### Description of the bug

* Exact model of your **printer**: Monoprice MSP. 
* **ADVi3++** exact version (Mainboard): 4.0.3 with BLTouch. 
* Printing using **SD card** or **USB**: SD card. 
* **For USB printing only**: Software used to print with its exact version:
* **Do you use a BLTouch?**: If yes, exact version of the BLTouch: V2.1
* Expected behaviour: Only the Z axis should move and the system should not lock up. 
* Actual behaviour: All 3 axis moves and the only way to stop the motion is to power off or push the reset button. 
* Steps to reproduce:
  - Start a print from an SD card. 
  - Wait until the the printing begins. 
  - Press the stop button twice. 
  - Wait until the motion stops. 
  - Go to the Move menu and press the Z up arrow.

andrivet, 2020-01-14:
I am able to reproduce this behavior. Thank you for the precise bug report,

Salty-Doug, 2020-01-14:
You are welcome.
I’m glad to help with the code.  I really like using it.

andrivet, 2020-01-20:
Fixed in 4.0.5

closed 2020-01-20 by andrivet

Salty-Doug, 2020-01-21:
@andrivet When do you plan on building 4.0.5?

commit 0d2d6af9 (2020-01-20): Do not disconnect the host; Clear command queue to fix #219

Move version to 4.0.5-dev

Change the place of the debugging code

Adjust Pause and Stop code