andrivet/ADVi3pp #222: Support for M80/M81 for Balco 3D Printer (Mainboard 5.2C) / MKII

BogdanTheGeek, 2019-12-08:
Thank you for submitting your feedback to the ADVi3++ project.

Please be sure to review our [code of conduct](code_of_conduct.md) and the checklists below before submitting and use one of the templates. 

**It is mandatory**. If you do not comply with these rules, I will **delete** your issue or comment. **I will be very strict**. If you persist, I may **ban** you.

- [x] This issue (or comment) is **NOT** about support. **This is not the right place to ask for support**. 
Instead, go to [ADVI3++ Community - Get Help](https://community.advi3pp.com/c/get-help). 

------------------------------------------------------------
### Feature Request

- [x] Your request is not specific to your setup but can benefit to many people
- [x] You have checked that you are not creating a **duplicate** of an existing (maybe closed) request (use the search)
- [x] You have provided a short but meaningful title
- [x] You have provided enough information (such as links to web sites, blogs, technical description) so that anyone can understand what you are talking about. Just any only a title is not enough.
- [x] You have provided objective **facts**, not your personal feelings, your personal preferences.

#### Detailed description of the feature request:
Marlin has support for Power on and off wit M80/M81 and if the printer is always connected to OctoPrint then the main board remains powered up even if the main PSU is disconnected. This is useful as it makes the printer silent. I see most of the code is there to enable this to work and M81 returns ok but it doesn't do anything.
What is left to do in order to implement this feature. I can do it myself, I just need a bit of guidance from someone who knows the code better.

andrivet, 2020-02-16:
Information posted on ADVi3++ Community:
https://community.advi3pp.com/t/printer-auto-shutdown/233/28

Relay board goes to pin #17 (PH5)

[image]

[image]

andrivet, 2020-04-24:
Information posted on ADVi3++ Community:
https://community.advi3pp.com/t/aldi-balco-he180021-5-2c-pcb/

> I checked the printer internals and came to the conclusion that the additional power switch pcb had developed a fault (rather coincidentally at the same time as updating the firmware) so I have simply bypassed it.
>
> This board (photo attached with modified bypass wiring) appears to produce an initial power on activation and relay energisation for the power supply activation, but then seems to require a signal, either continuous or pulsed in the form of a watchdog from the mainboard (2560 pin 17 PH5/OC4C) via a 2pin cable and into an opto isolator to maintain the power supply on.
>
> Printer now powers up ok.

[image]

MGuy7, 2020-12-23:
Wow read this  tried out as, no click on power up, Yep it doe's what you said on the Can... sweet Thanks...
As this board is a mains sub system switch based on the main MotherBoard via a watchDog presence, however I have to 
remberer not to leave the printer unsupervised, as no supervisory electronics is now in use....
Thanks again Michael........

andrivet, 2020-12-23:
@MGuy7 what are you saying? Is it some kind of spam? Can you make sentences?

MGuy7, 2020-12-28:
Sorry if not correct posting, please delete remove if you haven't already..
I have a balco 3D printer that wouldn't turn on,
Searching for info found this post, which made sense to me, and after doing
the suggested, the printer powered up fine. Looking to get the BL Touch
probe and put your firmware on and reuse the printer.. again many apologies
if this if it didn't read correctly, I was very happy that the printer is
now working..

King regards Michael..

andrivet, 2024-12-25:
Implemented in version 6.0

closed 2024-12-25 by andrivet