andrivet/ADVi3pp #223: Disabling beeps does not work with some LCD displays

andrivet, 2019-12-30:
### Bug Report

- [x] You have checked that you are not creating a **duplicate** of an existing (maybe closed) issue (use the search)
- [x] You have tested with the **latest** stable version of ADVi3++
- [x] You are able to reproduce the bug
- [x] You are able to describe step by step the process to reproduce the bug

#### Description of the bug

* Exact model of your **printer**: Aldi Belgium Balco 3D Printer Touch
* **ADVi3++** exact version (Mainboard): 4.0.3
* Printing using **SD card** or **USB**: n/a
* **For USB printing only**: n/a
* **Do you use a BLTouch?**: n/a
* Expected behaviour: It should be possible to disable beeps with the LCD panel
* Actual behaviour: It does not. And if beeps are activated, you get in fact double beeps.
* Steps to reproduce: See this discussion https://community.advi3pp.com/t/beep-toggle-option-double-beep/

andrivet, 2020-01-18:
I think this is fixed with 4.0.5. Hard to say for sure since I do not have such a buggy LCD screen. I will reopen this ticket if necessary.

jacobcboe, 2020-02-07:
I also have this issue - installed 4.0.5 yesterday.
Exact model of your printer: Monoprice Maker Select Plus (US)
ADVi3++ exact version (Mainboard): 4.0.5
Printing using SD card or USB: n/a
For USB printing only: n/a
Do you use a BLTouch?: No
Expected behaviour: It should be possible to disable beeps with the LCD panel
Actual behaviour: It does not. And if beeps are activated, you get in fact double beeps.
Steps to reproduce: See this discussion https://community.advi3pp.com/t/beep-toggle-option-double-beep/

Picture of LCD Board:
[image]

Picture of Version Info:
[image]

andrivet, 2020-02-07:
Reopen since it is not solved. I have however not enough information to be able to fix it in the near future.

jacobcboe, 2020-02-07:
Understood - no problem. Muting is nice to have, but definitely not essential. Let me know if there is any other information I can gather to help out.

andrivet, 2020-02-08:
@jacobcboe It is not essential, but these beeps are so annoying.

The information I need requires some hardware material (USB to TTL) and some knowledge of those displays. One solution would be to get one of these displays. I have already this in the past for another problem (I bought a display from one of the user of ADVi3++). I will contact you on [ADVi3++ Community](https://community.advi3pp.com) to see what we can do.

andrivet, 2020-02-10:
I have ordered a new panel from AliExpress and I will exchange it with one of the problematic panels (thanks @jacobcboe for your help). I hope to be able to find a solution. However, the panels have to cross the oceans two times (from Singapore to USA and then from USA to Switzerland) so it will take some weeks.

j0hnm4r5, 2020-04-14:
I'm on a Monoprice Maker Select Plus; the LCD always beeps (even with the toggle turned off), but does a weird double beep (almost as if it's trying to play two sounds at once) when the toggle is turned on.

j0hnm4r5, 2020-04-14:
I too have this exact problem, except with a Monoprice Maker Select Plus.

andrivet, 2020-07-12:
Followup on this issue: I have found a solution. Will be part of version 5.0.

Gymy20, 2021-01-03:
Hello. I have the same problem here with wanhao i3 plus mark 1. I've installed the last version. The beeps are not deactivated!

andrivet, 2021-09-03:
Probably fixed in version 5 but keep this open until it is confirmed.

closed 2021-09-18 by andrivet