andrivet/ADVi3pp #227: 4.0.4 Auto bed leveling stuck at homing

Hunt3red, 2020-01-08:
### Bug Report

- [X] You have checked that you are not creating a **duplicate** of an existing (maybe closed) issue (use the search)
- [X] You have tested with the **latest** stable version of ADVi3++
- [X] You are able to reproduce the bug
- [X] You are able to describe step by step the process to reproduce the bug

#### Description of the bug

* Exact model of your printer: Wanhao Duplicator i3 Plus Mark II
* ADVi3++ exact version (Mainboard): 4.0.4
* Do you use a BLTouch?**: No
* Expected behaviour: Measuring positions and returning measuring results on a grid.
* Actual behaviour: When starting automatic bed leveling, the LCD says homing... and the printer is homing, but when it seemingly has finished homing, the printer just stops moving and the LCD still says homing...
* Steps to reproduce:
  - Go to controls, start automatic leveling
* Additional info: Downgrading to 4.0.3 makes the feature work again.

andrivet, 2020-01-08:
Can anybody owning a Mark II confirm this behavior?

andrivet, 2020-01-08:
Maybe related to #226

andrivet, 2020-01-18:
This should be fixed in 4.0.5 since I believe it related to #226.

closed 2020-01-18 by andrivet