andrivet/ADVi3pp #237: Enable Manual Mesh Bed Leveling

jallen810, 2020-03-21:
Andrivet,

I just installed the latest revision on my MPMaker Select. One feature I would very much like to see is the option to enable manual mesh bed leveling. I saw some other people talking about this but have not found the raw files to try and accomplish this myself and don't see anything other than the basic screw nut leveling in the existing firmware. 
1.) I would very much like to see this added to the "Tuning" menu in your next firmware with the option to customize the number of points to probe in the X & Y AND allow an offset for binder clips
2.) Can you perhaps point me in the direction the source code for your firmware and basic instructions? Note: I have watched Teaching Tech's [video](https://www.youtube.com/watch?v=lHLMkjnuJyU) on how to do this with an older build of yours so I have a rough idea what I am doing. My only real question is how to make the machine raise/lower the Z at each point without having a button on-screen? Could Pronterface/S3D control be used instead?

andrivet, 2020-03-22:
Why not using a BLTouch. It is far superior and well supported by ADVi3++.

jallen810, 2020-03-22:
Because BLtouch are expensive and finicky at best, probing works fine on my other machines and is free.

 

Jake Allen 
PH: 314-660-9100
EM: ja11en@icloud.com

Mark 9:23 God Bless

andrivet, 2020-03-22:
https://community.advi3pp.com/t/manual-mesh-bed-leveling/

andrivet, 2020-07-30:
Looks mostly like a duplicate of #59.
Regarding "option to customize the number of points to probe in the X & Y", I will not implement this. Should be done at compile time. Too hacky to do this at runtime time.

closed 2020-07-30 by andrivet