andrivet/ADVi3pp #244: In the Print Settings Page, make the actual Z height visible above the baby carriage (Pram)

Salty-Doug, 2020-06-08:
Thank you for submitting your feedback to the ADVi3++ project.

Please be sure to review our [code of conduct](code_of_conduct.md) and the checklists below before submitting and use one of the templates. 

**It is mandatory**. If you do not comply with these rules, I will **delete** your issue or comment. **I will be very strict**. If you persist, I may **ban** you.

- [ ] This issue (or comment) is **NOT** about support. **This is not the right place to ask for support**. 
Instead, go to [ADVI3++ Community - Get Help](https://community.advi3pp.com/c/get-help). 

------------------------------------------------------------
### Feature Request

- [ x] Your request is not specific to your setup but can benefit to many people
- [x ] You have checked that you are not creating a **duplicate** of an existing (maybe closed) request (use the search)
- [x ] You have provided a short but meaningful title
- [x ] You have provided enough information (such as links to web sites, blogs, technical description) so that anyone can understand what you are talking about. Just any only a title is not enough.
- [x ] You have provided objective **facts**, not your personal feelings, your personal preferences.

#### Detailed description of the feature request:

After extruder maintenance which disturbs the BLTouch Z-Height (offset), good first layer adhesion is sometimes difficult to achieve. 
In the Print Settings, accessible by pressing the gear symbol in the Print page, there are several tools available to make adjustments during printing. There is a Baby Steps Tool to adjust the nozzle height. This is very useful, however the Z height display in the header does not show the real height as adjusted with the Baby Steps tool.  the header only shows the commanded Z height from the GCODE file.
Could the actual Z height be displayed above, as part of the Baby Steps Tool?

------------------------------------------------------------
#### Description of the Environment

* Exact model of your **printer**: MMSelect Plus
* **ADVi3++** exact version (Mainboard): 4.0.6
* Printing using **SD card** or **USB**: OctoPrint and SD card.
* **For USB printing only**: Software used to print with its exact version:
* **Do you use a BLTouch?**: Yes, version 3.1
* Expected behaviour: Display the exact Z height as adjusted by the Baby Steps Tool.
* Actual behaviour:  The only Z height information in the header is the Z height as set in the GCODE file.
* Steps to reproduce:
  - Start a print.
  - Observe the deposition of the first layer and adjust the Z height to achieve a perfect squish of the first layer.

andrivet, 2020-06-08:
Note: I have made some reformatting of your message.

Something more or less like that is planned for version 5.

Salty-Doug, 2020-06-10:
I noticed that this issue is a duplicate of [213](https://github.com/andrivet/ADVi3pp/issues/213). 
I will close this issue

closed 2020-06-10 by Salty-Doug