andrivet/ADVi3pp #245: Automatic nozzle cleaning at start of a print.

Tjoeben, 2020-07-09:
Thank you for submitting your feedback to the ADVi3++ project.

Please be sure to review our [code of conduct](code_of_conduct.md) and the checklists below before submitting and use one of the templates. 

**It is mandatory**. If you do not comply with these rules, I will **delete** your issue or comment. **I will be very strict**. If you persist, I may **ban** you.

- [x] This issue (or comment) is **NOT** about support. **This is not the right place to ask for support**. 
Instead, go to [ADVI3++ Community - Get Help](https://community.advi3pp.com/c/get-help). 

------------------------------------------------------------
### Feature Request

- [x] Your request is not specific to your setup but can benefit to many people
- [x] You have checked that you are not creating a **duplicate** of an existing (maybe closed) request (use the search)
- [x] You have provided a short but meaningful title
- [x] You have provided enough information (such as links to web sites, blogs, technical description) so that anyone can understand what you are talking about. Just any only a title is not enough.
- [x] You have provided objective **facts**, not your personal feelings, your personal preferences.

#### Detailed description of the feature request:

In Marlin there is the G12 code for Clean the Nozzle.
Is it possible to implement this in the new ADVi3pp ?

This is not really necessary for PLA. But for PET-G, it is really helpful.
When this can be configured from the user interface so that everyone can choose where to place the brush.

A few links:
https://marlinfw.org/docs/gcode/G012.html
https://www.youtube.com/watch?v=jF4MmRxuiG0

------------------------------------------------------------
### Bug Report

- [ ] You have checked that you are not creating a **duplicate** of an existing (maybe closed) issue (use the search)
- [ ] You have tested with the **latest** stable version of ADVi3++
- [ ] You are able to reproduce the bug
- [ ] You are able to describe step by step the process to reproduce the bug

#### Description of the bug

* Exact model of your **printer**:
* **ADVi3++** exact version (Mainboard):
* Printing using **SD card** or **USB**:
* **For USB printing only**: Software used to print with its exact version:
* **Do you use a BLTouch?**: If yes, exact version of the BLTouch
* Expected behaviour:
* Actual behaviour:
* Steps to reproduce:
  - Do this
  - Do that

andrivet, 2020-07-11:
To work, this requires a cleaning area. I do not see (mechanically) where such cleaning can be places. There is no accessible room for it. Or you already have an idea about that?

andrivet, 2020-07-30:
No answer to my questions so closing.

closed 2020-07-30 by andrivet