andrivet/ADVi3pp #246: Skew compensation implementation

Slug-Cat, 2020-07-17:
Feature request

I think a skew correction option in the GUI could be of great use for tuning. It was added to Marlin in 1.1.8. There are often either very slight or very noticeable misalignments in the axes, and skew correction can "fix" these misalignments. For me specifically, this would be incredibly useful for mechanical parts in which tolerances need to be as close as possible. One of my axes are askew because the actual screw hole used to affix the frame to the body is ever so slightly misaligned.

[https://marlinfw.org/docs/gcode/M852.html](https://marlinfw.org/docs/gcode/M852.html)

andrivet, 2020-07-17:
From https://imgur.com/a/1RbvOD9

[image]

[image]

https://forum.prusaprinters.org/forum/assembly-and-first-prints-troubleshooting/bed-skew/

https://marlinfw.org/docs/configuration/configuration.html#configuring-marlin
under "Bed Skew Compensation"

andrivet, 2022-06-18:
Implemented in version 5.4.0

closed 2022-06-18 by andrivet