andrivet/ADVi3pp #248: Compensation of X-axis twist when using BL-Touch - to solve initial layer differences on X-axis direction

MauroGil66, 2020-08-10:
### Feature Request

- [ ] Your request is not specific to your setup but can benefit to many people
- [ ] You have checked that you are not creating a **duplicate** of an existing (maybe closed) request (use the search)
- [ ] You have provided a short but meaningful title
- [ ] You have provided enough information (such as links to web sites, blogs, technical description) so that anyone can understand what you are talking about. Just any only a title is not enough.
- [ ] You have provided objective **facts**, not your personal feelings, your personal preferences.

#### Detailed description of the feature request:
**Background:** I've seen some guys (including myself) having problems with different layer thicknesses between the left and right positions on the bed when using BL-Touch, due to a small twist on X-axis. The problem goes worse as far as BL-Touch has a higher off-set from the nozzle on the Y-axis. 
Example of the problem: https://community.advi3pp.com/t/bltouch-one-side-too-high-how-do-i-square-the-printer/2076

**Request:** Small twist differences could be compensated if a Z-height calibration could be performed also on the left and right side positions (B and D) besides the regular position (5 - bed center), so that difference could be adjusted in the leveling setup considering an approximation to linear movement. It's in fact a spiral movement but, for a small twist, such approximation could be valid (needs confirmation).

[image]

andrivet, 2020-08-16:
Your feature request is somewhat vague.

>Small twist differences could be compensated if a Z-height calibration could be performed also on the left and right side positions (B and D) besides the regular position

Do you have any link to any article to support this?

> such approximation could be valid (needs confirmation)

I need more that assumption.

As I understand your request, it will be very complicated for regular users to understand this. And concretely, how do you this this working?

MauroGil66, 2020-08-18:
> Do you have any link to any article to support this?

It's not necessary any article. It's simple math. Maybe I was not clear enough. Let my try to rephrase it: if someone uses a 0,1mm feeler gauge, as usually anyone does and, let's say, get a Z-height of 1.50mm at position 5. She/he does the same procedure and uses the same feeler gauge to calculate Z-height at B and D positions, getting, for instance, 1,40mm and 1,60mm respectively.
It means that twist on X axle is producing a difference of +0,1 at left (B position) and -0,1mm at right (D position) all necessary adjustment is to compensate such difference the same way the heated bed level is compensated.

> I need more that assumption.

Science and life are made by several assumptions. Someday you made an assumption that could develop better firmware for 3D printers. That was what moved you forward. Don't misunderstand me, we make a lot of assumptions when we study some kinds of stuff, for instance, gravity equal to 10m/s², pi = 3, etc. It's intended to simplify calculations when small decimals don't matter. The idea of assumptions are to make life easier.

> As I understand your request, it will be very complicated for regular users to understand this. And concretely, how do you this this working?

Sometimes it's not necessary for regular users to understand the essentials of any solution. They just need a step by step solution to get rid of their problems. I can show some people facing the same problem on your website, one of them was mine:
https://community.advi3pp.com/t/bed-leveling-is-incorrect/1993
https://community.advi3pp.com/t/bltouch-not-leveling-correctly/1399
https://community.advi3pp.com/t/bltouch-leveling-differences-from-left-to-right-side-of-heated-bed/1968
Look what a small difference of around a half-millimeter produced in my printer.
Those links above cope with the first checkpoint: 
- [X] Your request is not specific to your setup but can benefit to many people

The solution proposed in the following video is very difficult to get the necessary precision since there is no fine-tuning. It's difficult to get a lever with decimal precision. 
https://youtu.be/HNapPacc6Ic

I did it in my printer, but still get small differences:
[image]

I'm also modeling it in Fusion 360, but I still need some time to analyze the details. My model is available to anyone interested in taking a look at it: https://a360.co/3kWluwz
[image]
[image]

For the time being, a simple solution could be only a calculation of plus and minus as I've explained in my first sentence. Sometimes it's easier to go to trial and error than keep trying to create a Ph.D. thesis. As soon as I get more details, I will share it with you. Maybe someone could find a better way to explain what I'm trying to show here since I'm limited to what I have exposed above. With some CAD drawing maybe I can figure out a better way to explain the math and geometry behind the solution I'm proposing.  

All the best and keep safe!

andrivet, 2020-08-18:
Sorry, it is still not clear:

> It means that twist on X axle is producing a difference of +0,1

Or it means that the bed is actually not level. How do you know it is a twist and not the bed?

> Sometimes it's not necessary for regular users to understand the essentials of any solution

I was talking about the UI for the user, not of the underneath implementation. 

> They just need a step by step solution to get rid of their problems.

Which step by step solution?

MauroGil66, 2020-08-18:
> Sorry, it is still not clear:

Yes, I do understanding that it is a complex issue. It would be better if I could draw it. I'll try to do it tonight.

> Or it means that the bed is actually not level. How do you know it is a twist and not the bed?

We should know that the bed is not causing it because the BL-Touch should have corrected it. The Z-height on points B and D should be checked after a previous automatic leveling. The check on those two points is intended only for twist checking. If they match the same measure as in point "5" (bed center point), e.g. 0,1mm, they should be fine and no twist is present at all. Otherwise, it should be compensated in a certain way, adding or subtracting the differences from the previous measurements.
Another assumption is that we have done all checks proposed by you on this post:
https://community.advi3pp.com/t/the-left-side-is-too-close-the-right-side-is-too-high-or-the-contrary/100
and the problem still persists (that was my case).

> I was talking about the UI for the user, not of the underneath implementation.

It could be under Leveling:
[image]
I'll provide a suggestion of UI to you.
 
> Which step by step solution?

It could be under, e.g., X-Twist check:
Instruction example: "Use the same feeler gauge you used on point '5' to calibrate points B and D with the same friction intensity you have done in point '5'. If you get the same measurements on all 3 points, your X-Axis is ok. Otherwise, a compensation will be provided by the software."
That is the point where it is necessary to check if a simple addition or subtraction to previous automatic leveling values will fix the problem.
I'll keep providing updates on this topic.

andrivet, 2020-08-18:
It is more clear now, I will think about it.

Salty-Doug, 2020-08-18:
I have had this exact problem with my first MPMSP printer. I tried loosening the anchor screws and twisting the rods and it helped. I replaced the rods with new ones and it helped even more. 
Thanks for reporting this issue.

MauroGil66, 2020-08-19:
So, as promised, I've made some drawings in order to provide an easier understanding of the problem.

[image]

It's schematic but provides a better view of what happens. Of course, it's with exaggerated displacements for better visualization.
Blue line represents BL-Touch and red line the hot end. Since all leveling is provided by BL-Touch, the consequence is a hot end going down in one side of the heated bed and up at the other side.

andrivet, 2020-08-19:
Nice drawings and model in Fusion 360. Can you share the original of the drawings (your last post)? I will see if I can integrate them in the User Manual.

MauroGil66, 2020-08-19:
Yes, for sure! You can use them wherever you want. I did it in CorelDraw version 2019. I also exported it to AutoCad DWG.
[X-axis twist drawings.zip](https://github.com/andrivet/ADVi3pp/files/5096235/X-axis.twist.drawings.zip)
Please, let me know if you need any other format or modification of the sent ones. It will be my pleasure to provide any help.

andrivet, 2020-08-19:
Here is a draft of the screen:

[image]

and could be under Leveling:

[image]

MauroGil66, 2020-08-20:
It looks very nice. Congrats

andrivet, 2020-08-20:
Note: On the draft of the screen, multipliers are 0.01, 0.10, 1.00. 1.00 does not make much sense. I will change the values.

andrivet, 2020-08-25:
Well, after experimental implementation and testing, it is more complicated than expected. It is not correct to modify the mesh map because each time you send a G29 command, it will erase the X twist compensation values. So it means that I have to modify the code of Marlin itself. However, I think that it will provide a huge enhancement to ADVi3++ so I will implement it.

Salty-Doug, 2020-08-26:
One word in the X-Twist Compensation screen should be changed. 
The statement "Adjust L and R heights to be at the same position than M."
To be correct English it should be "Adjust L and R heights to be the same as M."

Thanks for working on this.

andrivet, 2020-08-26:
@Salty-Doug Thanks. As English is not my mother tongue, I am not always good as finding correct but short sentences.

andrivet, 2021-06-03:
Implemented in version 5

closed 2021-06-03 by andrivet