andrivet/ADVi3pp #251: Low temp trim feature

gonzob, 2021-04-08:
This is a feature request.

Currently my Cocoon Create shows 12C and 15C for the extruder and bed temps respectively when the room temp is 22C. I’ve checked the high temp readings and they are pretty good. This is not a problem, except during the winter when the room temperatures may drop to 10C. Under these conditions, the printer refuses to heat because the measured temps are below the 5C limit. I have to use a hot air gun to first heat the bed enough so it can heat, then the heat off the bed raises the temp of the extruder enough to heat.

I’d like a manual low temp trim feature for the room temp readings for both the bed and the extruder.

Gonzo

andrivet, 2021-04-08:
Your printer has a hardware problem (since the temperature readings are wrong). So this is not really a feature request. I will not implement a feature to fix a hardware problem specific to you printer.

gonzob, 2021-04-09:
Fair enough. I'll live with it. Close the issue.

andrivet, 2021-04-09:
But you should take care of this problem. Do not hesitate to post on ADVi3++ Community (https://community.advi3pp.com/c/printers/).

closed 2021-04-11 by andrivet