andrivet/ADVi3pp #253: Changing sorting of files on the SD card

andrivet, 2021-09-12:
### Bug Report

- [x] You have checked that you are not creating a **duplicate** of an existing (maybe closed) issue (use the search)
- [x] You have tested with the **latest** stable version of ADVi3++
- [x] You are able to reproduce the bug
- [x] You are able to describe step by step the process to reproduce the bug

#### Description of the bug

* Exact model of your **printer**: Wanhao Duplicator i3+
* **ADVi3++** exact version (Mainboard): 5.0.0
* Printing using **SD card** or **USB**: SD
* Expected behaviour: files should be sorted by creation/update time, most recent first
* Actual behaviour: The files are sorted in alphabetic order
* Steps to reproduce:
  - Put some files on the SD card
  - Access then with Print.

ianw, 2021-09-12:
I've looked a bit more closely and I think the files with 5.0.0 are actually sorted in chronological order (i.e. oldest first), not alphabetical.  I note that ``Marlin/Configuration_adv.h`` by default seems to have this setup to do what you'd expect (show latest thing copied first ... assuming you don't delete or otherwise fiddle files; it's not actually sorting the timestamp as such but the FAT order)

```
  // Reverse SD sort to show "more recent" files first, according to the card's FAT.
  // Since the FAT gets out of order with usage, SDCARD_SORT_ALPHA is recommended.
  #define SDCARD_RATHERRECENTFIRST
```

it doesn't seem ``SDCARD_SORT_ALPHA`` is enabled; it seems like that overrides this behaviour.  I've never formatted, or even deleted anything from this card, so I'm still a bit lost as to why it would have flipped around.

andrivet, 2021-09-18:
Duplicate of #95

closed 2021-09-18 by andrivet