andrivet/ADVi3pp #261: Diagonal movement during manual leveling

guyvia, 2021-09-24:
When adjusting the corners, it will move along x and then y to go corner to corner, instead of moving at a 45 degree angle. This is certainly not a huge issue, but it would be a quality of life improvement for those of us without auto-leveling.

andrivet, 2022-06-04:
Do not see the point of the issue. What's wrong with the current behaviour?

cjastone, 2022-06-07:
Having upgraded recently from 4.0.6, this used to be the old behaviour.  It's much faster when performing a series of small, iterative adjustments when trying to get the bed "juuust right".

I'm not sure with which version this behaviour was changed, however it has certainly increased the time taken for these movements.

andrivet, 2022-06-07:
Closing in favour of #289

closed 2022-06-07 by andrivet