andrivet/ADVi3pp #268: Z Height Bug

patrick505, 2021-12-14:
#### Description of the bug
I have discovered 2 potential bugs with Z height settings. When setting the Z height the value saved is .01 less than the value entered. When trying to change this value via settings>sensor>Z offset, the value does not save.

* Exact model of your **printer**: Maker Select Plus
* **ADVi3++** exact version (Mainboard): 5.01 BL touch
* Printing using **SD card** or **USB**: Both
* **For USB printing only**: Software used to print with its exact version: Octoprint 1.7.2
* **Do you use a BLTouch?**: If yes, the exact version of the BLTouch: 3.1
* Expected behavior:
Setting Z height via LCD, the actual value entered is saved. When changing this value via settings>sensor>Z offset, the new value is saved.
* Actual behavior:
When setting z-height via the LCD menu calibration, the value saved is .01 less than what was selected during the process (thus always an odd number). When trying to change this value via settings>sensor>Z offset, the value is not saved and the original value set during the z height calibration is still used.

* Steps to reproduce:
Set height via Leveling>Z height. Go through calibration, z height ends up .01 less
Go to settings>sensor>Z offset, enter a new value, click next. The printer reads "settings stored." Go back to the z offset setting and the old value remains.

andrivet, 2021-12-14:
I will investigate

andrivet, 2022-03-25:
So first bug was a rounding problem (float to int). Second bug was the wrong type used (unsigned instead of signed). Fixed in version 5.1.

closed 2022-03-25 by andrivet