andrivet/ADVi3pp #270: When attempting to compile 5.0.1 or 5.0.2, MarlinUI errors

DiskDoctor, 2022-01-03:
Truncated errors. I added a dummy alert_level in ultralcd.h for testing. Fixed the first error, but of course it wasn't valid and created other errors. 

Are the packages missing something, or outdated ultralcd.* or  ??

I read about earlier versions of ADVi3pp needing Arduino IDE 1.6, but nothing about the recent versions.

Please advise. I can offer to test or troubleshoot as requested. Non-BL version. Compiling only, no board attached.

Errors:
Arduino: 1.8.19 (Windows Store 1.8.57.0) (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Files\Arduino\3D Printer\ADVi3pp-advi3-5.0.1\ADVi3pp-advi3-5.0.1\Marlin\src\lcd\ultralcd.cpp:66:21: error: 'uint8_t MarlinUI::alert_level' is not a static data member of 'class MarlinUI'

   uint8_t MarlinUI::alert_level; // = 0

C:\Files\Arduino\3D Printer\ADVi3pp-advi3-5.0.1\ADVi3pp-advi3-5.0.1\Marlin\src\lcd\ultralcd.cpp:67:55: error: 'char MarlinUI::status_message [64]' is not a static data member of 'class MarlinUI'

   char MarlinUI::status_message[MAX_MESSAGE_LENGTH + 1];

andrivet, 2022-01-03:
I just tried compiling both sources (by downloading them, so starting from scratch) and everything is fine. It does compile without any error. So I don't know exactly what you are doing (since you gave almost no details), but the problem is on your side.

```
...
Compiling .pioenvs/advi3pp_51/FrameworkArduino/wiring_digital.c.o
Compiling .pioenvs/advi3pp_51/FrameworkArduino/wiring_pulse.S.o
Compiling .pioenvs/advi3pp_51/FrameworkArduino/wiring_pulse.c.o
Compiling .pioenvs/advi3pp_51/FrameworkArduino/wiring_shift.c.o
Archiving .pioenvs/advi3pp_51/libFrameworkArduino.a
Indexing .pioenvs/advi3pp_51/libFrameworkArduino.a
Linking .pioenvs/advi3pp_51/firmware.elf
Checking size .pioenvs/advi3pp_51/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [======    ]  60.2% (used 4932 bytes from 8192 bytes)
Flash: [=====     ]  50.9% (used 129214 bytes from 253952 bytes)
Building .pioenvs/advi3pp_51/firmware.hex
================================================================= [SUCCESS] Took 17.82 seconds =================================================================

Environment    Status    Duration
-------------  --------  ------------
advi3pp_51     SUCCESS   00:00:17.816
```

closed 2022-01-03 by andrivet

DiskDoctor, 2022-02-08:
I listed some of the errors. Pretty clear. Also showed Marlin compiles fine, so not the environment. Are you changing the code somehow that isn't compiling properly? 
'uint8_t MarlinUI::alert_level' is not a static data member of 'class MarlinUI'  is pretty straightforward.

DiskDoctor, 2022-02-08:
In any case, it isn't usual to simply close an issue without feedback or questions/comments.

andrivet, 2022-02-08:
I have checked and the firmware is compiling fine from the sources here on GitHub, without any change. So the problem is on your side. I do not provide support on GitHub, that's why I have closed the issue (since it is not an issue).

andrivet, 2022-02-17:
Your title is not very good because it suggests that ADVi3++ does not compile at all. And this is why I do not see any issue and close it. So instead, what you want is to compile with arduino ide.

In the documentation of ADVi3++, I clearly said “Note: It may be possible to build ADVi3++ with Arduino IDE but I do not support it”

So again, issues here are not for support. This is not the right place. So please stop with your messages.

DiskDoctor, 2022-02-17:
No need to be snippy about it. 

Your software is NOT 100% compliant with Arduino standards and does not compile in Arduino. That is a fact. That is not any users' fault. That is your "mistake" or possibly intentional so you can end-run profiting from Open Source Software?

You make the software that runs on Arduino but will not compile on Arduino, but for a monthly subscription fee you offer to pre-compile the software using an alternative method.

All this written based on OPEN SOURCE software, which prohibits charges for derivative software. 

Charging for "support" is walking a thin line. Charging for support because you didn't write compliant software correctly...that's getting on shaky ground.

Telling those who reveal this issue with your code to 'stop posting' because you don't want to hear about your issues is NOT compliant with either the terms or spirit of GitHub and Open Source software.

You made some desirable additions to Marlin software. But if you try to close it off or shoo away Github users commenting on the lack of proper coding standards, you could lose it, or worse- be found VIOLATING the licensing terms of MARLIN and other included code.

Strongly suggest you take a different Attitude and different approach toward fellow programmers who reveal issues and problems with your code.

DiskDoctor, 2022-02-17:
And note my issue clearly states the issue was found when compiling with Arduino IDE.