andrivet/ADVi3pp #274: Enhanced motion during filament load

galfisk, 2022-03-10:
Thank you for submitting your feedback to the ADVi3++ project.

Please be sure to review our [code of conduct](code_of_conduct.md) and the checklists below before submitting and use one of the templates. 

**It is mandatory**. If you do not comply with these rules, I will **delete** your issue or comment. **I will be very strict**. If you persist, I may **ban** you.

- [X] This issue (or comment) is **NOT** about support. **This is not the right place to ask for support**. 
Instead, go to [ADVI3++ Community - Get Help](https://community.advi3pp.com/c/get-help). 

------------------------------------------------------------
### Feature Request

- [X] Your request is not specific to your setup but can benefit to many people
- [X] You have checked that you are not creating a **duplicate** of an existing (maybe closed) request (use the search)
- [X] You have provided a short but meaningful title
- [X] You have provided enough information (such as links to web sites, blogs, technical description) so that anyone can understand what you are talking about. Just any only a title is not enough.
- [X] You have provided objective **facts**, not your personal feelings, your personal preferences.

#### Detailed description of the feature request:

When filament load starts, the extruder is lifted, which is good. When it finishes, the extruder is immediately lowered again. If it started out very close to the print bed (or a printed object), this jams the nozzle into the recently purged filament, making a mess. I suggest either not lowering the extruder afterwards, or reminding the user to clear the purged material and pressing Next to lower the nozzle.

------------------------------------------------------------

galfisk, 2022-03-10:
It would also be nice if the preparatory Z lift is done when tuning the extruder, since that entails even more purging of filament.

andrivet, 2022-03-10:
I think that a better solution is to have a proper stopping gcode to move the nozzle away. 

> reminding the user to clear the purged material

frankly, I don't want to go in this direction. I do not like systems that are always asking if you are sure you want to do this or that.

>It would also be nice if the preparatory Z lift is done when tuning the extruder, since that entails even more purging of filament.

No idea of what you are taking about.

galfisk, 2022-03-10:
I'm talking about this: 

When going to Controls->Add filament, the extruder is automatically raised some centimeters to make space for purged filament.

It would be nice if, when going to Tuning->Extruder, the extruder was also automatically raised to make space for purged filament, since extruder tuning entails purging even more filament.

andrivet, 2022-03-10:
OK, this is another request about a different part of the firmware. It is always better to create on topic per request, i will do so.

galfisk, 2022-03-10:
Ok, I'll keep that in mind. Thanks.

closed 2022-06-29 by andrivet