andrivet/ADVi3pp #275: Manual move controls not respecting limit switches

galfisk, 2022-03-10:
Thank you for submitting your feedback to the ADVi3++ project.

Please be sure to review our [code of conduct](code_of_conduct.md) and the checklists below before submitting and use one of the templates. 

**It is mandatory**. If you do not comply with these rules, I will **delete** your issue or comment. **I will be very strict**. If you persist, I may **ban** you.

- [X] This issue (or comment) is **NOT** about support. **This is not the right place to ask for support**. 
Instead, go to [ADVI3++ Community - Get Help](https://community.advi3pp.com/c/get-help). 

------------------------------------------------------------
### Bug Report

- [X] You have checked that you are not creating a **duplicate** of an existing (maybe closed) issue (use the search)
- [X] You have tested with the **latest** stable version of ADVi3++
- [X] You are able to reproduce the bug
- [X] You are able to describe step by step the process to reproduce the bug

#### Description of the bug

* Exact model of your **printer**: Wanhao Duplicator i3 Plus (5 machines total, bug tested and confirmed on two)
* **ADVi3++** exact version (Mainboard): 5.0.3 (on mainboard 5.2c) and 5.0.1 (on mainboard 5.1).
* Printing using **SD card** or **USB**: SD card
* **Do you use a BLTouch?**: no
* Expected behaviour: when pressing either control to jog in the X, Y or Z direction, I expect the motion to stop when hitting the limit switches, or reaching the mechanically allowed extent of motion away from the limit switches
* Actual behaviour: the motion continues, making the motors grind. The only motion that stops without grinding is raising the gantry in the Z direction.
* Steps to reproduce:
- Turn on printer
- Go to Controls-Move
- Move X left, Y rear or Z down to engage limit switch
- Motor doesn't stop when switch is engaged
- Move X right or Y front
- Motor doesn't stop when mechanical limit is reached (no limit switch in this direction)
- Move Z up
- This time the motors do stop before the gantry hits the top of the printer.

The limit switches work normally during other operations, such as homing, bed leveling and printing.

galfisk, 2022-03-10:
Update: after I started and subsequently aborted a print, the movement controls started respecting the limit switches. After I turned the printer off and on, it started ignoring them again. This is on the printer with 5.0.3 (5.2c)

andrivet, 2022-03-24:
I am not able to reproduce this kind of behavior and never heard of something like that by other people. Are you sure it is not a (hardware) problem with your printer? Like a bad connection somewhere? I am closing the issue for the moment (until it is confirmed by someone else)

closed 2022-03-24 by andrivet