andrivet/ADVi3pp #276: Some but not all files with the letter "å" in the file name immediately go to 100% completion

galfisk, 2022-03-10:
Thank you for submitting your feedback to the ADVi3++ project.

Please be sure to review our [code of conduct](code_of_conduct.md) and the checklists below before submitting and use one of the templates. 

**It is mandatory**. If you do not comply with these rules, I will **delete** your issue or comment. **I will be very strict**. If you persist, I may **ban** you.

- [X] This issue (or comment) is **NOT** about support. **This is not the right place to ask for support**. 
Instead, go to [ADVI3++ Community - Get Help](https://community.advi3pp.com/c/get-help). 

------------------------------------------------------------
### Bug Report

- [X] You have checked that you are not creating a **duplicate** of an existing (maybe closed) issue (use the search)
- [X] You have tested with the **latest** stable version of ADVi3++
- [X] You are able to reproduce the bug
- [X] You are able to describe step by step the process to reproduce the bug

#### Description of the bug

* Exact model of your **printer**: Wanhao Duplicator i3 Plus
* **ADVi3++** exact version (Mainboard): 5.0.3 (5.2c)
* Printing using **SD card** or **USB**: SD card
* **Do you use a BLTouch?**: No
* Expected behaviour: file should print
* Actual behaviour: printer immediately says print is 100% completed, without doing anything
* Steps to reproduce:
- Make a GCode file named "WDI3P_Fibox hållare testbit 2 full.gcode"
- Try printing it from SD card
- Pritner immediately says the progress is 100%
- Rename file to "WDI3P_Fibox hallare testbit 2 full.gcode"
- Now it starts printing from 0%
- Not all files with å in them have this issue. Rename the file "WDI3P_Knapphållare v1.gcode" and it also works.

The printer cannot display the "å" character, and the letter after it disappears also. So it will show "WDI3P_Knapphållare v1.gcode" as "WDI3P_Knapphlare v1.gcode"

There may also be issues with the other Scandinavian letters (ä, ö, æ, ø, and the capitals Å, Ä, Ö, Æ, Ø). Worth checking out as well.

galfisk, 2022-03-10:
Update: this bug seems to change character depending on what's been done to the printer. After a bed leveling and some other stuff (can't recall exactly, but preheating and some jogging was also done), the "WDI3P_Knapphållare v1.gcode" file now goes straight to 100%. Removing and reinserting the SD card made no difference. Renaming the file to remove the "å" worked.

galfisk, 2022-03-10:
Hm. And after the last reboot of the printer, it wouldn't load the file wihtout the å either. When loading the file, the printer says the filename starts with a dot. Can't see that dot when opening it in the computer though. When I readded the original file with the å, it would print. I need to characterize this issue further.

andrivet, 2022-03-10:
I will look at this, but I think that the printer is not able to handle non-ASCII characters (both at the level of the LCD panel and at the Marlin level). FAT support in Marlin is pretty basic and it already took a lot of memory (both SRAM and Flash). Anyway, it will test and if indeed I can confirm it is not possible (or too complicated) to support, I will add this information in the Manual.

galfisk, 2022-03-10:
Yes, there's some other issue at hand, since the last time I tried, one of the files without "å" failed in the same way. I'm using this printer now, and students are using the others, so I can't continue troubleshooting at the moment, but I'll do some more testing tomorrow or monday.

(Btw, one of the teachers was permitted to use his work credit card to sponsor you for a year on Patreon, so that we can properly thank and reward you for your hard work.)

galfisk, 2022-03-11:
Hi again, it turns out that the actual problem is caused by something entirely different. I'll create a new issue for that. You can close this issue once you've decided what to do with the issue of non-ASCII characters.

andrivet, 2022-04-03:
After reviewing the possibilities of the LCD panel and of Marlin, I think it is too complicated to try to support Unicode filenames. So instead, non-ASCII characters (i.e < 128) will be replaced by _.
This will be part of version 5.2.

closed 2022-04-03 by andrivet