andrivet/ADVi3pp #278: Hide gcode files where the file name starts with a period

galfisk, 2022-03-11:
Thank you for submitting your feedback to the ADVi3++ project.

Please be sure to review our [code of conduct](code_of_conduct.md) and the checklists below before submitting and use one of the templates. 

**It is mandatory**. If you do not comply with these rules, I will **delete** your issue or comment. **I will be very strict**. If you persist, I may **ban** you.

- [X] This issue (or comment) is **NOT** about support. **This is not the right place to ask for support**. 
Instead, go to [ADVI3++ Community - Get Help](https://community.advi3pp.com/c/get-help). 

------------------------------------------------------------
### Bug Report

- [X] You have checked that you are not creating a **duplicate** of an existing (maybe closed) issue (use the search)
- [X] You have tested with the **latest** stable version of ADVi3++
- [X] You are able to reproduce the bug
- [X] You are able to describe step by step the process to reproduce the bug

#### Description of the bug

Created from issue #276. I thought the issue was caused by Nordic characters in the filename, but it was caused by the below behaviour instead, when using an iPad to move files from cloud storage to an SD card.

* Exact model of your **printer**: Wanhao Duplicator i3 Plus
* **ADVi3++** exact version (Mainboard): 5.0.3 (5.2c)
* Printing using **SD card** or **USB**: SD card
* **Do you use a BLTouch?**: no

File names starting with a dot are hidden files created by Unix-based OSes. When interacting with FAT32 memory cards, iPadOS (and presumably iOS, Mac OS and Linux) may create copies of .gcode files with the name  ._<filename>.gcode in order to store file properties that FAT32 can't handle.

* Expected behaviour: It would be useful if the printer would hide these files, since they're not printable
* Actual behaviour: The files are visible and can be selected for printing, causing confusion, as they're often the newest files on the card and therefore listed at the end
* 
* Steps to reproduce:
- Add/move/rename files to a FAT32-formatted SD card using a Unix-based OS (tested with iPadOS, may be the same with iOS, MacOS and Linux)
- The OS sometimes automatically creates junk .gcode files with similar file names as existing files, but starting with a dot and underscore.
- The printer lets the user select these files for printing, but since they contain no gcode, the printer immediately says that printing is 100% complete.

Fix: hide files starting with a dot. A leading dot is a Unix naming convention for a hidden file.
Improvement: show an error, rather than "100%", if a file is empty or contains something else than gcode

andrivet, 2022-03-11:
Note: This is a behavior specific of macOS, iOS and iPadOS. Linux and Unixes do not create such dot files along with regular files. 

It is actually not a bug. The printer let you select what is on the card. If you put a JPEG file on the SD card, it will appear and let you select it. I am assuming in ADVi3++ that people do know what they are doing.

You are in a special context since you are not operating yourself the printer but have students that do. I am not sure I want to complexify ADVi3++ just for this special context.

galfisk, 2022-03-11:
Fair enough. We'll work around it.

andrivet, 2022-03-11:
I want to think more about it. So I reopen it. I will change it from bug report to enhancement request.

galfisk, 2022-03-11:
Okay, thanks.

andrivet, 2022-04-03:
Implemented in version 5.2. Marlin is ignoring hidden files and most of files starting with a period are hidden. But for whatever reason, in your case, some are not. So ignore them based on the long file name (not the short one).

closed 2022-04-03 by andrivet